Thom-x/docker-fr24feed-piaware-dump1090 #4: Disable webserver logging for dump1090

SaturnusDJ, 2019-06-07:
When reading the logs from Docker, it is full with lines like this:
`[07/Jun/2019 22:54:46] "GET ..............`
Those are webserver logs from dump1090.
Any way to disable webserver logs?

Thom-x, 2019-06-09:
I think you can modify `supervisord.conf` and change the log output.

```
[program:python]
command=/usr/bin/python -m SimpleHTTPServer 8080 
directory=/usr/lib/fr24/public_html
stdout_logfile=/dev/null
stdout_logfile_maxbytes=0
stderr_logfile=/dev/null
stderr_logfile_maxbytes=0
```

SaturnusDJ, 2019-06-11:
Thanks that worked.

Because no vi text editor is present I copied the original file from terminal, edited it and injected it back via echo.

closed 2019-06-16 by Thom-x