Thom-x/docker-fr24feed-piaware-dump1090 #6: mac address

Technikte, 2019-06-16:
I have read this and wonder which mac address ? The one of my network adapter or ?
Change --mac-address="ff:ff:ff:ff:ff:ff" with your own MAC address.

I am running the docker container on unraid with an usb dvt stick.
The fr24 side is working good but since the update today the local skyview is loading endless.

Thom-x, 2019-06-16:
The mac adresse is here to work as an id for piaware. It's not related to
the local web site not loading.

Thom-x, 2019-06-16:
Can you please give the error and your configuration ?

Technikte, 2019-06-16:
Sure, where to get the correct id to use with the docker container ?

> PiAware - Claim and Link a Brand New PiAware Ground Station
> Success!
> You claimed the following 1 receivers:
> 
>     xx1xxxx4-1e20-xxxx-bd84-5ecxxxxa5xx7

Is this the correct one ? Would be --mac-address="xx1xxxx4-1e20-xxxx-bd84-5ecxxxxa5xx7" then correct ?

Thom-x, 2019-06-16:
Oh I see, you need to put a new parameter in piaware.conf.
I'll find it and give you the solution as soon as possible. 😉

Technikte, 2019-06-16:
> 
> 
> Oh I see, you need to put a new parameter in piaware.conf.
> I'll find it and give you the solution as soon as possible. 😉

Alright, please update the documentation for FlightAware and docker run command in the readme, too.
Would be to nice to know if need to include mac-adress and if I need to alternate the config for fr24 and/or config.js. Thanks for the help.

Thom-x, 2019-06-16:
I updated the README, we now have to use `feeder-id xxxxxxxx` found under "Unique Identifier" in flightaware website.

3d40895f41144b04d9d6fd35e1eb82d33973bc99

closed 2019-06-16 by Thom-x