Thom-x/docker-fr24feed-piaware-dump1090 #7: WebUI of dump not working

Technikte, 2019-06-16:
I moved this problem to a new issue. This is only true with new docker container using 
`Sun Jun 16 15:35:56 2019 CEST dump1090-fa starting up.`

I am running this container on unraid without "mac-address" parameter and with empty FlightAware login data (not using fa) and filled fr24feed.ini

Flightradar24 is working good and I can reach and see everything on the "Flightradar24 Feeder/Decoder" url.
When I try to reach the webui on port 8080 the layout with time and date loads, only the map is not loading and in the midlle the loading circle. I used the fresh config.js without any mods.

I see in the logs of the docker container follow beside the normal feed stuff..

> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET / HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /style.css?v=2 HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /jquery/jquery-ui-1.11.4-smoothness.css HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /jquery/jquery-3.0.0.min.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /jquery/jquery-ui-1.11.4.min.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /jquery/plugins/jquery.validate.min.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /ol/ol-4.4.2.css HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /ol/ol-4.4.2.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /registrations.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /planeObject.js?v=2 HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /ol/ol3-layerswitcher.css HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /layers.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /script.js?v=2 HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /ol/ol3-layerswitcher.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /config.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /markers.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /dbloader.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /formatter.js?v=2 HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /flags.js HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /spinny.gif HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/icon-information@2x.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/fa_logo_color.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/pa-sv-logo.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/ff-sv-logo.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/settings-icon.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/map-icon.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/table-icon.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/alt_legend_feet.svg HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /images/toggle-width@2x.png HTTP/1.1" 200 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] code 404, message File not found
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /status.json HTTP/1.1" 404 -
> 192.168.178.50 - - [16/Jun/2019 15:36:20] code 404, message File not found
> 192.168.178.50 - - [16/Jun/2019 15:36:20] "GET /favicon.ico HTTP/1.1" 404 -
> 

Not quite sure if this is caused because I am not using fa.

Thom-x, 2019-06-16:
I forgot to change config.js, the new one (from flightaware) has quite a bite changed.
Sorry about that, I think it will be fine now.

424b6e78590d572c0ec9162483f9b54ca3eb41f4

Technikte, 2019-06-16:
> 
> 
> I forgot to change config.js, the new one (from flightaware) has quite a bite changed.
> Sorry about that, I think it will be fine now.
> 
> [424b6e7](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/commit/424b6e78590d572c0ec9162483f9b54ca3eb41f4)

Thanks its working now after updating the container and all files.

As a followup for https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/6
`2019-06-16 15:36:48Z warning: /etc/piaware.conf:9: invalid value for option feeder-id: YOUR_FEEDER_ID`
Since I did not use fa I have not changed anything in piaware.conf..so the default value should be somehting like 0000 or so instead of YOUR_FEEDER_ID.

closed 2019-06-16 by Technikte

Thom-x, 2019-06-16:
It's someting like the example.