Thom-x/docker-fr24feed-piaware-dump1090 #13: support dump1090 --net-only

jkldgoefgkljefogeg, 2020-02-18:
I want to use this with network input (receiver on a different device sending tcp raw/beast output) to this container

Thom-x, 2020-03-02:
Yes you can, for that you need to change the configuration like a0a2b3ec8aee0ff658281425a78d394aa33b0e5c

closed 2020-03-02 by Thom-x

jkldgoefgkljefogeg, 2020-04-09:
My receiver is over network and supports beast/raw/SBS output over different ports. Looks like piaware isn't working correctly

```
$ docker run -d -p 8080:8080 -p 8754:8754 \
> -e "FR24FEED_FR24KEY=***" \
> -e "FR24FEED_RECEIVER=beast-tcp" \
> -e "FR24FEED_HOST=192.0.2.61:32457" \
> -e "PIAWARE_FEEDER_DASH_ID=***" \
> -e "PIAWARE_RECEIVER_DASH_TYPE=relay" \
> -e "PIAWARE_RECEIVER_DASH_HOST=192.0.2.61" \
> -e "PIAWARE_RECEIVER_DASH_PORT=32457" \
> -e "HTML_SITE_LAT=31.55601" \
> -e "HTML_SITE_LON=152.264783" \
> -e "HTML_SITE_NAME=test" \
> -e "PANORAMA_ID=D6M76JGN" \
> thomx/fr24feed-piaware
```

> $ docker logs dreamy_mclaren
> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
> [s6-init] ensuring user provided files have correct perms...exited 0.
> [fix-attrs.d] applying ownership & permissions fixes...
> [fix-attrs.d] done.
> [cont-init.d] executing container initialization scripts...
> [cont-init.d] 01-confd-fr24feed: executing... 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[235]: INFO Backend set to env
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[235]: INFO Starting confd
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[235]: INFO Backend source(s) set to 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[235]: INFO Target config /etc/fr24feed.ini out of sync
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[235]: INFO Target config /etc/fr24feed.ini has been updated
> [cont-init.d] 01-confd-fr24feed: exited 0.
> [cont-init.d] 02-confd-piaware: executing... 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[248]: INFO Backend set to env
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[248]: INFO Starting confd
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[248]: INFO Backend source(s) set to 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[248]: INFO Target config /etc/piaware.conf out of sync
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[248]: INFO Target config /etc/piaware.conf has been updated
> [cont-init.d] 02-confd-piaware: exited 0.
> [cont-init.d] 03-confd-html: executing... 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[260]: INFO Backend set to env
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[260]: INFO Starting confd
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[260]: INFO Backend source(s) set to 
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[260]: INFO Target config /usr/lib/fr24/public_html/config.js out of sync
> 2020-04-09T06:04:30Z 22cd420795bb /opt/confd/bin/confd[260]: INFO Target config /usr/lib/fr24/public_html/config.js has been updated
> [cont-init.d] 03-confd-html: exited 0.
> [cont-init.d] 04-upintheair: executing... 
> --2020-04-09 06:04:30--  http://www.heywhatsthat.com/api/upintheair.json?id=PAR5DWMG&refraction=0.25&alts=1000,10000
> Resolving www.heywhatsthat.com (www.heywhatsthat.com)... 69.39.100.105
> Connecting to www.heywhatsthat.com (www.heywhatsthat.com)|69.39.100.105|:80... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: unspecified [application/json]
> Saving to: '/usr/lib/fr24/public_html/upintheair.json'
> 
>      0K .......... .......... .....                             236K=0.1s
> 
> 2020-04-09 06:04:31 (236 KB/s) - '/usr/lib/fr24/public_html/upintheair.json' saved [26622]
> 
> [cont-init.d] 04-upintheair: exited 0.
> [cont-init.d] done.
> [services.d] starting services
> Thu Apr  9 06:04:31 2020 UTC  dump1090-fa v3.8.0 starting up.
> 2020-04-09 06:04:31 | [main][i]FR24 Feeder/Decoder
> 2020-04-09 06:04:31 | [main][i]Version: 1.0.18-5/generic
> 2020-04-09 06:04:31 | [main][i]Built on Mar  4 2016 16:26:55 (devel-d11ca48.git/Linux/x86_64)
> 2020-04-09 06:04:31 | [main][i]Copyright 2012-2016 Flightradar24 AB
> 2020-04-09 06:04:31 | [main][i]http://flightradar24.com
> rtlsdr: no supported devices found.
> [services.d] done.
> s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
> s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
[... 8 lines cut ...]
> 2020-04-09 06:04:31 | [reader][i]Connected to the receiver, configuring
> 2020-04-09 06:04:31 | [reader][i]Configured, processing messages
> [cont-finish.d] done.
> [s6-finish] waiting for services.
> 2020-04-09 06:04:31 | [httpd][i]Server started, listening on 0.0.0.0:8754
> 2020-04-09 06:04:31 | [master][i]Starting processing thread
> 2020-04-09 06:04:31 | [mlat][i]Waiting for MLAT configuration
> ****************************************************
> piaware version 3.8.0 is running, process ID 314
> your system info is: Linux 22cd420795bb 5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 GNU/Linux
> 2020-04-09 06:04:31 | [reader][w]Setting new UTC offset: 0!
> [s6-finish] syncing disks.
> [s6-finish] sending all processes the TERM signal.
> 2020-04-09 06:04:31 | [time][i]Synchronizing time via NTP
> 2020-04-09 06:04:31 | [main][i]Terminating on user request
> Reloading configuration and reconnecting.
> piaware (process 314) is shutting down because it received a shutdown signal (SIGTERM) from the system...
> piaware (process 314) is exiting...
> 2020-04-09 06:04:31 | [reader][i]Connection terminated
> 2020-04-09 06:04:31 | [reader][i]Terminating on request
> [s6-finish] sending all processes the KILL signal and exiting.
> 2020-04-09 06:04:34 | [time][i]Time synchronized correctly, offset +0.0008 seconds
> 2020-04-09 06:04:34 | [main][i]Feed Network client started
> 2020-04-09 06:04:34 | [main][i]Terminating worker threads
> 2020-04-09 06:04:34 | [feed][x]Feeding thread terminated

jkldgoefgkljefogeg, 2020-04-09:
How does this docker-compose serve one feed to different consumers (fr24feed, piaware)? Does it use dump1090-fa to receive 1 feed, then locally serve fr24feed and piware? or fr24feed/piaware each uses 1 instance of dump1090 to connect to server?

Thom-x, 2020-04-09:
It starts dump1090 locally and then fr24 and piaware connect to it.

If your dump1090 is on a other network you need to disable the container from starting it.

`-e SERVICE_ENABLE_DUMP1090=false`

Thom-x, 2020-04-09:
In your case dump1090 cannot start because there is no receiver, crash, and all services are terminated.

jkldgoefgkljefogeg, 2020-04-09:
Thanks. I got fr24feed to work. Does piaware work without dump1090? Mine is showing problem fetching data from dump1090

```
172.17.0.1 - - [09/Apr/2020 06:39:08] "GET /data/aircraft.json?_=1586414307178 HTTP/1.1" 404 -
no ADS-B data program seen listening on port 30005 for 190 seconds, next check in 60s
```

Thom-x, 2020-04-09:
Both fr24 and piaware are working but the website cannot without dump1090 locally.

jkldgoefgkljefogeg, 2020-04-09:
In my setup (getting beast output over TCP from network), is there any difference between PIAWARE_RECEIVER_DASH_TYPE other vs relay (it's not clear to me from piaware documentation)

https://flightaware.com/adsb/piaware/advanced_configuration

Is there a way to confirm piaware is working? https://flightaware.com/adsb/stats/user/ worked after switching type from relay to other. I am not sure if it's just a delay from the website.

Thom-x, 2020-04-09:
I read that relay is for an installation with sdcard (not the setup we are using), I think it would also work but nothing will be different. You can tell me if it's the case.

Also I don't really now how to check if piaware is working expect from going to the page that you mentioned and check the "last updated" on the top of the receiver.

jkldgoefgkljefogeg, 2020-04-09:
https://flightaware.com/adsb/stats/user/ Site Configuration shows Log at the bottom of the tab

Looks like _other_ receiver type works, relay does not (0 msg received)

> [2020-04-08 23:49 PDT] 15 msgs recv'd from the ADS-B data program at 192.0.2.61/32457 (15 in last 5m); 15 msgs sent to FlightAware

This tool is awesome! Thanks so much.

It would be neat to be support dump1090 over network as well (piaware web UI, dump1090 web UI/json)

Thom-x, 2020-04-09:
Ok thanks.
I'll take a look at making it works with a remote dump1090. 😉

Thom-x, 2020-04-09:
The easy way to do it is to update `/usr/lib/fr24/public_html/script.js` (bind mount it) and change every `data/xxx.json` with `http://yourremotedump1090/data/xxx.json`, but you need to need your remote dump1090 to have the webserver running.

jkldgoefgkljefogeg, 2020-04-09:
I don't have a remote dump1090. The remote connection is to a hardware ADS-B receiver with network export support.

Thom-x, 2020-04-09:
Oh my bad ok.
In this case we need to do it the hard way with something like that : https://github.com/wiedehopf/combine1090

Thom-x, 2020-04-09:
@jkldgoefgkljefogeg 
I added the possibility to pass custom args to dump1090, give it a try.
https://hub.docker.com/layers/thomx/fr24feed-piaware/dump1090-custom-config/images/sha256-ff964851f2442e42351249ede1051dd4ef1574c24dc6f4e323907b3d259f446e?context=repo
https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/tree/dump1090-custom-config

jkldgoefgkljefogeg, 2020-04-09:
are args for DUMP1090_ADDITIONAL_ARGS separated by space, like `-e "DUMP1090_ADDITIONAL_ARGS=--net-only --net-bi-port 33200 --net-bo-port 33201" \` ?

This way only 1 connection to remote ADB-B receiver is needed and piaware/fr24feeder can read from local dump1090 in docker?

dump1090 doesn't support TCP client (actively connect to ads-b server). nc is needed to set up TCP connection to remote receiver over network and forward received packets to dump1090 listener port.

`nc -d 192.0.2.61 32457 | nc 127.0.0.1 33200`. 

probably needs another dump1090 envvar to run (restart) nc to connect to ads-b server

Thom-x, 2020-04-09:
> are args for DUMP1090_ADDITIONAL_ARGS separated by space, like -e "DUMP1090_ADDITIONAL_ARGS=--net-only --net-bi-port 33200 --net-bo-port 33201" \ ?

Yes this should works like that.

Thom-x, 2020-04-09:
You can try to run the container with a custom `CMD` : 

```docker run ... thomx/fr24feed-piaware /bin/bash -c "apt-get update && apt-get install -y netcat && nc 192.0.2.61 32457 | nc 127.0.0.1 33200"```

jkldgoefgkljefogeg, 2020-04-09:
I see why SkyAware doesn't have any data. SkyAware is part of dump1090-fa served by a HTTP server (I thought the web interface was served directly by dump1090-fa). Without dump1090 running, there is nothing updating SkyAware data and only static UI is served.

https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/dump1090-custom-config/root/etc/services.d/http/run

Thom-x, 2020-04-09:
Yes exactly..

vistalba, 2020-09-08:
Should work now with rtl_tcp and the image with https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/29#issuecomment-688724345