Thom-x/docker-fr24feed-piaware-dump1090 #15: Some error but working site

Technikte, 2020-02-29:
Since the update today (29.02) it seems to be working all good but I have seen the following errors since some time already.

```
2020-02-29 14:13:21,321 INFO Included extra file "/etc/supervisor/conf.d/supervisord.conf" during parsing
2020-02-29 14:13:21,329 INFO RPC interface 'supervisor' initialized
2020-02-29 14:13:21,329 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-02-29 14:13:21,330 INFO supervisord started with pid 6
2020-02-29 14:13:22,332 INFO spawned: 'python' with pid 9
2020-02-29 14:13:22,332 INFO spawned: 'fr24feed' with pid 10
2020-02-29 14:13:22,333 INFO spawned: 'piaware' with pid 11
error | Local time: 2020-02-29 14:13:22 +0100
error | GMT+0 time: 2020-02-29 13:13:22 +0100
error | Your machine should be set as GMT+0 time zone!
warning | Time zone is not set to GMT+0
2020-02-29 14:13:22 | [main][i]FR24 Feeder/Decoder
2020-02-29 14:13:22 | [main][i]Version: 1.0.24-5/generic
2020-02-29 14:13:22 | [main][i]Built on Oct 30 2019 07:23:36 (static-i386_amd64-builds-and-macos-dump1090-update-c308dae.git/Linux/amd64)
2020-02-29 14:13:22 | [main][i]Running on: debian10
2020-02-29 14:13:22 | [main][i]Local IP(s): 172.17.0.2
2020-02-29 14:13:22 | [main][i]Copyright 2012-2019 Flightradar24 AB
2020-02-29 14:13:22 | [main][i]https://www.flightradar24.com
2020-02-29 14:13:22 | [main][i]DNS mode: LIBC
2020-02-29 14:13:22 | [main][i]Automatic updates are DISABLED
2020-02-29 14:13:22 | ERROR: bash: /sbin/rmmod: No such file or directory
2020-02-29 14:13:22 | [w]Detected --net argument for dump1090, disabling internal RAW feed!
2020-02-29 14:13:22 | [w]Detected --net argument for dump1090, disabling internal BS feed!
2020-02-29 14:13:22 | info | [httpd]Server started, listening on 0.0.0.0:8754
2020-02-29 13:13:22Z warning: /etc/piaware.conf:7: unrecognized option flightaware-user
2020-02-29 13:13:22Z warning: /etc/piaware.conf:8: unrecognized option flightaware-password
2020-02-29 14:13:22 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
2020-02-29 14:13:22 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
2020-02-29 14:13:22 | [main][i]Reader thread started
```

I would ignore the error with the different false time zone, but this "bash error" could be maybe fixed and "piaware.conf:7: unrecognized option" seems to be a problem since I got my piaware.conf like this, which should correct.
```
# This file configures piaware and related software.
# You can edit it directly or use piaware-config from the command line
# to view and change settings.
#
# If /boot/piaware-config.txt also exists, then settings present in
# that file will override settings in this file.
flightaware-user user123
flightaware-password password123456
feeder-id b***b645-****-****-****-d59ab6f***3c
```

The flightaware feeder seems to be working when I visit the "My ADS-B" page.

Thom-x, 2020-02-29:
I got the same error for the configuration, you can remove username and password it's deprecated.

I do not have the time error.

And it looks like it missing a library for rmmod

Thom-x, 2020-03-01:
fixed 66ce1feb66d2126f57ac2cf09a71e341ebb8ccf8

closed 2020-03-01 by Thom-x