Thom-x/docker-fr24feed-piaware-dump1090 #20: Change port 8080

sarkyscouser, 2020-03-23:
Hi I have this (mostly) working, but my server already has port 8080 allocated to another service that I don't want to mess with.

In fr24 feeder settings I've tried adding:

`--net-http-port 8081`

and adjusting the docker ports but that doesn't work either.

Is there a trick with the ports mapped to the container, or perhaps something I could add to the environment variables to make the map available on a different port please?

Port 8080 is after all a commonly used port so not a good choice by default.

Thom-x, 2020-03-23:
You can do docker run -p 6666:8080 to bind with another port on the host.

sarkyscouser, 2020-03-23:
I've already tried that as well.  I don't think it works because port 8080 is already being used by another docker container so it doesn't matter which port is mapped to 8080 I always get diverted to the existing service.

There needs to be another way of changing the map port. Is there a html environment variable that could do this?

Thom-x, 2020-03-23:
I don't understand why there is a conflict.
The port 8080 is internal to the docker container.
If you bind it to 6666 there is no way it can conflict with another one. on the host.

sarkyscouser, 2020-03-23:
Actually now that I think about it again that's a very good point.  Must be some other reason this is happening...

Happy for this to be closed whilst I track down the real issue, thanks.

closed 2020-03-23 by sarkyscouser