Thom-x/docker-fr24feed-piaware-dump1090 #21: Log says device or resource busy, although aircrafts are tracked

simonszu, 2020-03-26:
I have this issue since around 2 weeks. The log output shows the following lines repeatedly:
```
2020-03-26 07:28:54 | [reader][i]Connected to the receiver, configuring
2020-03-26 07:28:54 | [reader][i]Configured, processing messages
Thu Mar 26 07:28:54 2020 UTC  dump1090-fa v3.8.0 starting up.
rtlsdr: using device #0: Generic RTL2832U (Generic, RTL2832U, SN 77771111153705700)
usb_claim_interface error -6
```
However, connecting to the web frontend shows up some aircraft, although with type "Other" and not "ADS-B", whatever "Other" means. As a result of all this, my account on fr24 was degraded to Basic, since apparently my feeder is connected but i do not send any data. I have already double-checked whether some "real" DVB-T modules in my kernel are disabled, they are all blocked. 

How is the feeder currently tracking the aircrafts? Why isn't it able to connect to the stick all of a sudden?

Thom-x, 2020-04-05:
Are you using the latest version  of the image ? We had the same problem but it's now fixed.
See #14.

glowfishDE, 2020-07-01:
Have you set your parameters via the Docker Export variables OR the /etc/fr24feed.ini file correctly ?

If you use the ini File instead the variables, your ini File should probably look like this:

receiver="beast-tcp"                                                                                                                                                                                                                                                                              
host="127.0.0.1:30005"
fr24key="YOURFEEDERID"                                                                                                                                                                                                                                                                   
bs="yes"                                                                                                                                                                                                                                                                                     
raw="yes"                                                                                                                                                                                                                                                                                    
logmode="1"                                                                                                                                                                                                                                                                                  
procargs="--net --write-json /usr/lib/fr24/public_html/data"

If receiver "beast-tcp" is not working try "avr-tcp". You probably have "dvbt" in there which is not possible since your RTL SDR Stick is already in use (that is what the error message "usb_claim_interface error -6" is telling you.

So instead of trying to access the USB stick via the USB hardware, try to poll the data via the TCP connection.

Port 30005 is for Outbound Data in BEAST Mode to feed to fr24 and flightaware

closed 2020-12-25 by stale[bot]