Thom-x/docker-fr24feed-piaware-dump1090 #23: Update causes device busy

AustinShalit, 2020-05-11:
## Specifications

I just attempted an upgrade from `fa493ee73552d4ecdea6085aa46dac454a6468f837176aa1fe80fe894cb4e07c` to latest (`e6a44d31fdd73dd71fb9c433ddf9dc7e70a69eeb2009a0acd5f33a53a414a917`). I did this by stoping the old container, pulling the latest image, and then trying to start the container back up using the new image. This leads to no aircraft being tracked and the following messages repeated in the logs.

## Expected Behavior
Image is able to connect to USB stick.

## Actual Behavior
```
2020-05-11 09:17:51 | [reader][i]Connecting to Beast receiver via (exe:///usr/lib/fr24/dump1090 --net --write-json /usr/lib/fr24/public_html/data --raw --mlat)
2020-05-11 09:17:51 | [reader][i]Connected to the receiver, configuring
2020-05-11 09:17:51 | [reader][i]Configured, processing messages
Mon May 11 09:17:51 2020 UTC  dump1090-fa v3.8.1 starting up.
rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001197)
usb_claim_interface error -6
rtlsdr: error opening the RTLSDR device: Device or resource busy
2020-05-11 09:17:51 | [reader][i]Connection terminated
2020-05-11 09:17:51 | [main][i]Terminating child process 347 with SIGETERM
```

Thom-x, 2020-05-11:
Can you try to disable mlat ?
We already had this issue #14

AustinShalit, 2020-05-12:
How do I do that? I tried changing the following options to both be "no".
```
mlat="yes"
mlat-without-gps="yes"
```

Thom-x, 2020-07-01:
Do you mount the config file or are you using the environment var ?
I recommend you to use the env var, so no need to bind mount any file.
With env you only need to change your credentials (not mlat or anything else).

glowfishDE, 2020-07-01:
I had the same issue and got it working with setting the /etc/fr24feed.ini like this:

receiver="beast-tcp"                                                                                                                                                                                                                                                                              
host="127.0.0.1:30005"
fr24key="myfeederID"                                                                                                                                                                                                                                                                   
bs="yes"                                                                                                                                                                                                                                                                                     
raw="yes"                                                                                                                                                                                                                                                                                    
logmode="1"                                                                                                                                                                                                                                                                                  
procargs="--net --write-json /usr/lib/fr24/public_html/data"

Before i had the receiver set to "dvbt". When i changed it to "beast-tcp" (or avr-tcp) it was suddenly working without the mentioned error.

glowfishDE, 2020-07-01:
> Do you mount the config file or are you using the environment var ?
> I recommend you to use the env var, so no need to bind mount any file.
> With env you only need to change your credentials (not mlat or anything else).

I tried that before but it seems the export variables for fr24feed.ini were not recognized. So i had to use the mounted file instead :(

The export variables for piaware.conf were working howsoever.

Thom-x, 2020-07-01:
Strange I have something like this and it works, maybe you have more options:
```
    environment:
      - "HTML_SITE_LAT=47.xxxxxx"
      - "HTML_SITE_LON=-1.xxxxxx"
      - "HTML_SITE_NAME=HOME"
      - "FR24FEED_FR24KEY=xxxxxx"
      - "PIAWARE_FEEDER_DASH_ID=xxxxxxx"
```

closed 2020-12-25 by stale[bot]