Thom-x/docker-fr24feed-piaware-dump1090 #25: Location in Skyview displayed incorrect

glowfishDE, 2020-07-01:
Hi,

in the Skyview (http....:8080) my feeders location and center of the map is shown a few hundred meters offset. 

I have the following docker export variables that should give the correct location:

HTML_SITE_LON 11.123456
HTML_SITE_LAT 48.123456

I have set my coordinates also in flightaware and fr24 and restarted everything multiple times. When i change the export variables, the map and center spot are moving accordingly. But it seems its never displayed at the exact spot i set. 

The location is displayed slightly offset (~ 700 meters away from my house, on a totally different building).

Are the export variables cut to less than 6 digits after the dot ?

Thom-x, 2020-07-01:
You can try to view what's in the javascript config file:

[image]

glowfishDE, 2020-07-01:
Oh yes.. I forgot to mention. I have it set there too. Just double checked my set of coordinates. They point to the correct spot but the skyview shows it offset :(

Thom-x, 2020-07-01:
You don't need to change it manually in `config.json` this is the direct view from chrome.

At startup coordinates are added into the file automatically, **don't forget double quote**:
```
-e "HTML_SITE_LAT=11.123456"
-e "HTML_SITE_LAT=48.123456"
```

glowfishDE, 2020-07-01:
I have it also in the export variables .. Since i use a synology NAS for the docker container it looks like this:

[image]

It seems to obey these settings as my map and center spot is moving when i change the variables. However it´s never displayed at the exact right spot :(

Thom-x, 2020-07-01:
Are the coordinates correct when you show the dev console on your browser like in my screenshot ?

glowfishDE, 2020-07-01:
I checked in chromes developer console - yes its the correct coordinates in config.json ..also in the env variables its set the same. i just checked the dockerfile.

really strange

Thom-x, 2020-07-01:
In this case it's not an issue with the docker image.
It's a bug in piaware frontend I think.

glowfishDE, 2020-07-01:
Who is maintaining the frontend of piaware ? Would that be https://github.com/flightaware/piaware ?

Thom-x, 2020-07-01:
Yes I think.

glowfishDE, 2020-07-01:
Ok - i´ve opened an issue at the piaware github page. Closing this issue to avoid redundancy. Thanks for the support :)

closed 2020-07-01 by glowfishDE

glowfishDE, 2020-07-01:
It seems to be that dump1090 is rounding the GPS coordinate digits for privacy reasons.

You can set it manually from:

JSON_OPTIONS="--json-location-accuracy 1" (estimated)
to
JSON_OPTIONS="--json-location-accuracy 2" (exact position)

@Thom-x I just dont know yet how your docker setup is working. Do you have any idea where we can configure that variable? maybe as a docker env variable ?

Thom-x, 2020-07-01:
We start dump1090 [here](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/7ad29dc1acd60cf38d190d6f2e73843438142df8/root/etc/services.d/dump1090/run#L14), we could add `--json-location-accuracy 2` at the end.

glowfishDE, 2020-07-01:
Could we make it a docker env flag / variable so users can set it without modifying files inside the docker container and break their upgrade path ? That would be pretty cool :)

Thom-x, 2020-07-01:
It's possible, we can do something like `HTML_SITE_LAT` env var.

glowfishDE, 2020-07-01:
> It's possible, we can do something like `HTML_SITE_LAT` env var.

Cool.. i suggest HTML_SITE_LOCATION_ACCURACY <1|2|3>

Thom-x, 2020-07-01:
I created on more generic one, see #26

Thom-x, 2020-07-01:
You can try with image version `thomx/fr24feed-piaware:dump1090-additional-args`

glowfishDE, 2020-07-01:
> I created on more generic one, see #26

Ah very cool. When will this change go into the "latest" image ? Or is there a way to re-install a docker container with a different release type ?

My docker piaware is built with the latest branch. Sorry - i´m kinda new to docker :(

And if you pull a new image into your container, it will have a new feeder ID right? or can i set the old feeder ID somewhere to the new installed software somewhere ?

Thom-x, 2020-07-02:
If you change the version `latest` by `dump1090-additional-args` you don't have to do anything else.
It will work right away.