Thom-x/docker-fr24feed-piaware-dump1090 #28: Error when starting container

robertscheib, 2020-07-15:
## Specifications

  - Image hash or version: 
  - Platform: Raspberry Pi 3
  - Subsystem: Raspberry Pi OS

## Expected Behavior
FR24Feed and piaware to run and begin sharing

## Actual Behavior

Container starts up and then shuts down. Portainer is showing the following error in the log.  It repeats for every attempt to restart, it includes nothing else.

standard_init_linux.go:211: exec user process caused "exec format error"

I have included what I believe to be necessary screenshots. 

[image]
[image]
[image]
[image]

## Steps to Reproduce the Problem

  1. Start container in Portainer
  2. Monitor start progress
  3. Check log for additional error reference (show above in "Actual Behavior"

Thom-x, 2020-07-15:
You cannot run this image version with a Raspberry Pi 3, you need to build it yourself.

With docker hub it's not possible to build images for arm architecture. The only solution right now is to build it yourself.

`docker build -t myimagename .`

closed 2020-07-15 by Thom-x

e2jk, 2021-02-28:
@robertscheib FYI thomx is apparently now pushing his images to the Docker Hub not only for `linux/amd64`, but also `linux/arm/v6` and `linux/arm/v7` - so I've been running this image on a Raspberry Pi 1B (yes, old ;) ) without issues or having to compile it myself (which would have been awfully slow).

robertscheib, 2021-02-28:
> @robertscheib FYI thomx is apparently now pushing his images to the Docker Hub not only for `linux/amd64`, but also `linux/arm/v6` and `linux/arm/v7` - so I've been running this image on a Raspberry [...]

Awesome, thanks for the update. I have to rebuild my Pi as the SS card just died.