Thom-x/docker-fr24feed-piaware-dump1090 #29: SkyAware doesn't show aircrafts when dump1090 is running not on localhost

vistalba, 2020-09-07:
## Specifications

  - Image hash or version: latest
  - Platform: kubernetes
  - Subsystem: ubuntu 18.04 LTS

## Expected Behavior
Enabled is HTTP, FR24FEED and PIAWARE.
Conrigured as beast-tcp is a remote IP of my LAN.
FR24FEED and PIAWARE is running fine and does report aircrafts to the web.

I can access HTTP service from my browser but there is no aircrafts on the Map and the List.

In the log I can find the following:
```
10.32.0.1 - - [07/Sep/2020 13:01:46] "GET /data/aircraft.json?_=1599476498563 HTTP/1.1" 404 -
10.32.0.1 - - [07/Sep/2020 13:01:47] code 404, message File not found
```

## Actual Behavior
SkyAware Map and list should show aircraft informations.

## Steps to Reproduce the Problem

  1. Run dump1090 an a different host in LAN (not localhost / not same docker container)
  2. Configure fr24feed-piaware-dump1090 docker image to use HTTP, fr24feed and piaware.
  3. Open WebGUI and try to show aircrafts on the map.

Thom-x, 2020-09-07:
Hello,

Running dump1090 on another host is not fully supported right now.
You should change the config file with something like https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/13#issuecomment-611425517.

vistalba, 2020-09-07:
Unfortunatly this is not possible as the docker container runs in kubernetes behind rProxy and device which dump1090 is running is not available from the remote network. So it isn't enought to just direct allow connection to the dump1090-device.

There is a other way that could work, but isn't possible with your container: https://github.com/antirez/dump1090/issues/87

run on dump1090-fa node:
`rtl_tcp -a 0.0.0.0 -f 1090000000 -s 2000000`

run inside the docker:
`nc 11.22.33.44 1234 | dump1090 --ifile -`

But this is not possible as I've to put `netcat` in front to pipe it :(
Would there be a way to make this possible in this image as a option?

I tried by myself but didn't get it to run... for me the image is crashing when I want to use dump1090 :( may because there is no RTL USB mapped to docker?

Thom-x, 2020-09-07:
>I tried by myself but didn't get it to run... for me the image is crashing when I want to use dump1090 :( may because there is no RTL USB mapped to docker?

Yes, I think you can override the configuration with `DUMP1090_ADDITIONAL_ARGS` with `--net-only` or something like that (see the doc).
But you still have to tweak a bit to use netcat.

I can't test it myself, I let you try it, and maybe open a pull request with your work.

vistalba, 2020-09-07:
I just played around and got it working.
I created a pull reqeust:
https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/pull/30

Important notes:

Requires appox. 35-40 Mbit/s bandwidth.
Example RTL_TCP command: `./rtl_tcp -a 0.0.0.0 -f 1090000000 -s 2400000 -p 30005 -P 28 -g -10`
Use correct frequency, sample rate and ppm_error value.

closed 2020-09-08 by Thom-x

Thom-x, 2020-09-08:
Thank you! Fixed by your PR #30 🙇

vistalba, 2020-09-08:
Testet latest docker image and it is working as expected.

vistalba, 2020-09-08:
Just one more thing.
As rtl_tcp is usinge to much bandwidth I looked out for another solution.
I found one that is working too.

You can run `dump1090-fa` on a remote host and configure `--net-bo-port=30005`. 
Then you need to configure extra arg `--net-bi-port=30004` on `dump1090` inside the docker container.
After that we can just netcat the the remote feed to the local dump1090 instande with `nc [REMOTE-DUMP1090] 30005 | nc -q1 localhost 30004`

I will try to get a PR for this too.

Thom-x, 2020-09-08:
That could be a good way to save bandwidth 👍
We just need to add a new "service" with `nc [NC-REMOTE-HOST] [NC-REMOTE-POST]| nc -q1 localhost [NC-LOCAL-PORT]`

vistalba, 2020-09-08:
> That could be a good way to save bandwidth 👍
> We just need to add a new "service" with `nc [NC-REMOTE-HOST] [NC-REMOTE-PORT]| nc -q1 localhost [NC-LOCAL-PORT]`

This is exactly  my idea... but I don't know how to solve it when we want to use multiple remote dump1090 instances.
So best way would be that we can specify multiple `NC-REMOTE-HOST` as `192.168.1.200,192.168.1.201,...` and it would netcat all of them to the localhost.

If we want to do this just for one, it should be easy.

Anyway... I think about rename the ENVs used.

RTL version to:
`RTL_TCP_REMOTE_HOST`
`RTL_TCP_REMOTE_PORT`
`RTL_TCP_OVER_NETCAT`

and for the new one we should use:
`DUMP1090_REMOTE_HOST`
`DUMP1090_REMOTE_PORT`
`SERVICE_ENABLE_DUMP1090_OVER_NETCAT`

Bandwidth with new solution looks like it is below 200kbps.

Thom-x, 2020-09-08:
We can rename args that way 👍 

Take a look at https://stackoverflow.com/questions/11917708/pipe-multiple-commands-into-a-single-command
Maybe we can pipe all `nc` into one ?
If it works, we need to find a way to split different host with `,` and it should be good.

vistalba, 2020-09-08:
Okay... I got it to work as a new netcat service but for now with just one remote dump1090 instance.

I've created a PR (https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/pull/31) so that others can use it too.

I renamed the ENVs to more match the used method. (RTL_TCP or DUMP1090).
I created a new ENV `SERVICE_ENABLE_IMPORT_OVER_NETCAT` and also updated readme to describe both cases in seperate topics.

New method with dump1090 is using less bandwidth. (appox. < 100 Kbit/s)

I hope that helps.

Thom-x, 2020-09-08:
>So best way would be that we can specify multiple NC-REMOTE-HOST as 192.168.1.200,192.168.1.201,... and it would netcat all of them to the localhost.

I'm not motivated enough to try to implement this one. But that's a good idea.

vistalba, 2020-09-08:
> > So best way would be that we can specify multiple NC-REMOTE-HOST as 192.168.1.200,192.168.1.201,... and it would netcat all of them to the localhost.
> 
> I'm not motivated enough to try to implement this one. But that's a good idea.

I do not have multiple rtl-sdr usb sticks so I don‘t know if I will do this too.

Thanks for your help! :-)

Thom-x, 2020-09-08:
Maybe multiple time with the same stick, or with a dummy blank endpoint.
Also, I've seen that it can create some trouble with MLAT etc, so it is not worth the trouble.

vistalba, 2020-09-09:
I just had the idea to specify it the other was like `host1:port1,host2:port2`

In this case we can just use this to split IPs and port by `,` and then replace `:` with space. The result is that each IP/Port combination is a separate line.

Example:
`echo "192.168.1.1:30001,192.168.1.20:30002,10.10.0.10:30003" |  awk -v RS=, '1' | sed 's/:/ /g'`

But I don't get it how I can make netcat connect to each of them simultaneously.

Thom-x, 2020-09-09:
>But I don't get it how I can make netcat connect to each of them simultaneously.

Me neither, maybe with the script of [combine1090](https://github.com/wiedehopf/combine1090/blob/master/combine1090.sh)

commit a4d1e50f (2020-09-08): feat: add support for rtl_tcp (#30)

fix #29
Requires appox. 35-40 Mbit/s bandwidth.
Example RTL_TCP command: ./rtl_tcp -a 0.0.0.0 -f 1090000000 -s 2400000 -p 30005 -P 28 -g -10
Use correct frequency, sample rate and ppm_error value.