Thom-x/docker-fr24feed-piaware-dump1090 #35: [BUG] Panorama not working with flightaware-v4.0 version

vistalba, 2020-11-30:
I just tested the new flightware-v4.0 taged docker image and panorama-ring isn't working anymore.

## Specifications

  - Image hash or version: flightaware-v4.0
  - Platform: kubernetes & docker
  - Subsystem: ubuntu 20.04

## Expected Behavior
Show panorama rings on map.

## Actual Behavior
Does not show the panorama rings after update to tag `flightaware-v4.0`.

## Steps to Reproduce the Problem

  1. Deploy Docker with panorama ringt `-e "PANORAMA_ID=ABCDFE"` use tag `latest` -> working
  2. Deploy Docker with panorama ringt `-e "PANORAMA_ID=ABCDFE"` use tag `flightaware-v4.0` -> not working

Thom-x, 2020-11-30:
It's a known bug we need to wait for dump1090 update
https://github.com/flightaware/dump1090/pull/82

Thom-x, 2021-03-23:
fixed by #48

closed 2021-03-23 by Thom-x

vistalba, 2021-03-29:
@Thom-x This is still not working for me with version 5.0.

The file `upintheair.json` is located at `/usr/lib/fr24/public_html/` but it doesn't display it in the browser.
I saw in the changelog (https://discussions.flightaware.com/t/announcing-piaware-5/74591 )that the file should be at `/usr/share/skyaware/html/` but this folder doesn't exist.

Any ideas why this doesn't work for me?

Thom-x, 2021-03-30:
The file should be in `/usr/lib/fr24/public_html/upintheair.json` it is working for me.
On this docker image the `skyaware` folder is copied at `/usr/lib/fr24/public_html/`

Thom-x, 2021-03-30:
Maybe clear the browser cache or directly go to the json location to see if it is here.

vistalba, 2021-03-30:
Strange... today morning it just works :-/ Sorry. For me it is now okay.