Thom-x/docker-fr24feed-piaware-dump1090 #37: Fr24feed won't start on Pi Zero

CalvinSchwartz, 2021-01-15:
## Specifications

  - Image hash or version: latest (~Week 51 2020)
  - Platform: Raspberry Pi Zero W
  - Subsystem: Raspbian Buster

## Expected Behavior
Inside the container fr24feed starts after it's called by "/fr24feed/fr24feed/fr24feed --signup"

## Actual Behavior
`bash: /fr24feed/fr24feed/fr24feed: No such file or directory`
The file fr24feed exists in the subfolder ans is executable but it wont start (I am no linux expert).
Also running the "complete" image fails because fr24feed fails to start. If fr24feed is disabled then dump1090 and piaware work like they should.

Building the image on a Pi Zero is not possible (failed after running for 4 hours).

## Steps to Reproduce the Problem
1. ```docker run -it --rm \
	-e "SERVICE_ENABLE_DUMP1090=false" \
	-e "SERVICE_ENABLE_HTTP=false" \
	-e "SERVICE_ENABLE_PIAWARE=false" \
	-e "SERVICE_ENABLE_FR24FEED=false" \
	thomx/fr24feed-piaware /bin/bash```
  2. run: `/fr24feed/fr24feed/fr24feed`

Thom-x, 2021-01-15:
Hello,

Can you please show me the output of `ls -la /fr24feed/fr24feef`.

I don't have a raspberry to test so it will be difficult to find what's wrong.

Thom-x, 2021-01-15:
Fr24feed is directly downloaded from flightradar website. Maybe it is not compatible with pi zero. Or some dependencies are missing.

CalvinSchwartz, 2021-01-15:
Sure, the output is: 
```
drwxr-xr-x 2 1000 1000    4096 Mar  4  2016 .
drwxr-xr-x 3 root root    4096 Dec 16 12:35 ..
-rw-r--r-- 1 1000 1000    1081 Mar  4  2016 LICENSE.fr24feed
-rwxr-xr-x 1 root root 1070228 Mar  4  2016 fr24feed
-rwxr-xr-x 1 1000 1000    8236 Mar  4  2016 timestamper
-rw-r--r-- 1 root root       9 Mar  4  2016 version.txt
```

If I cd into that folder the next error message is
`bash: fr24feed: command not found`
and I have no idea why it's not starting

CalvinSchwartz, 2021-01-17:
I did some additional checking:
- FR24feed runs perfectly fine if installed on Pi Zero
- Building your container on a Pi Zero seems to be impossible due to the low RAM and computing power.

Therefore I tried the [mikenye/docker-flightradar24](https://github.com/mikenye/docker-flightradar24) which also doesn't work. But if I edit the script slightly (see [Issue](https://github.com/mikenye/docker-flightradar24/issues/21)) it builds and runs perfectly fine using the precompiled .deb file from the [fr24 repo](https://repo-feed.flightradar24.com).

Therefore I guess building your Image on a Pi Zero will work but I can't confirm this because building the image would take hours.

Thom-x, 2021-01-17:
Could you try to re-download https://repo-feed.flightradar24.com/rpi_binaries/fr24feed_1.0.26-9_armhf.tgz extract it and start it directly from the container ?

It could be also good idea to try to download the .deb extract it and start it (like mikenye did).

Thom-x, 2021-01-17:
Can you show me the output of `uname -m`
Like here https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/a49850c58500fcc1135df1e79e8d5d55ad5d965f/build/s6-overlay.sh#L2 ?

Maybe the output is not armv6l or armv7l when building it with github action.

Thom-x, 2021-01-18:
Could you try with the latest image version `docker pull thomx/fr24feed-piaware:latest` ?

CalvinSchwartz, 2021-01-18:
Uname: arm6l
The [fr24feed_1.0.26-9](https://repo-feed.flightradar24.com/rpi_binaries/fr24feed_1.0.26-9_armhf.tgz) does not work but if I try the
[fr24feed_1.0.25-3](https://repo-feed.flightradar24.com/rpi_binaries/fr24feed_1.0.25-3_armhf.tgz) version it runs.
Therefore I guess your script itself is perfectly fine but the current fr24feed version simply doesn't run on a raspberry pi.
See: https://github.com/mikenye/docker-flightradar24/issues/14

Thom-x, 2021-01-18:
Good catch ! Maybe it is a good idea to downgrade to 1.0.25-3 till it is fixed.

Thom-x, 2021-01-18:
I forced version 1.0.25-3 for Rpi. You can check it with latest.

CalvinSchwartz, 2021-01-21:
Checked it and it works!

closed 2021-01-21 by CalvinSchwartz