Thom-x/docker-fr24feed-piaware-dump1090 #42: Update latest to v4.0 of flightaware/dump1090

e2jk, 2021-02-28:
## Specifications

  - Image hash or version: latest/1.3.1
  - Platform: Raspberry Pi 1B (old ;) ), armv6
  - Subsystem: ?

## Expected Behavior
I would expect the `latest` tag of this repo on the Docker Hub to use v4 of [flightaware/dump1090](https://github.com/flightaware/dump1090/releases/tag/v4.0).

## Actual Behavior
The image currently pushed (one month ago) to the Docker Hub with tag `latest` is [version 1.3.1](https://hub.docker.com/layers/thomx/fr24feed-piaware/1.3.1/images/sha256-4e09b38fad0db1079e387eccddf4414fc5fc67eb21996ec482be23b02f438ca1?context=explore) of this repo, which uses v3.8.1 of `flightaware/dump1090`.
There is an older (5 months ago) tagged [flightaware-v4.0](https://hub.docker.com/layers/thomx/fr24feed-piaware/flightaware-v4.0/images/sha256-031f66bb4cba42420b58a68edbee0ab20ed65e604d9f950b0ec1d583f7e330fe?context=explore) which, based on its name, I'm assuming is using v4 of `flightaware/dump1090`.

Shouldn't `latest` point to v4?

## Steps to Reproduce the Problem

  1. Install from Docker Hub targeting the `latest` tag
  2. Run and noticed that on the webpage it indicates v3.8.1 in the upper right of the page. I'd have expected v4.0

Thom-x, 2021-02-28:
Hello v4 has a bug #35, so I'm waiting for the fix (that is already on master, but no release yet...).

e2jk, 2021-02-28:
Merci @Thom-x.
I do see there are quite some changes on their dev branch ([86 commits since v4.0](https://github.com/flightaware/dump1090/compare/master...flightaware:dev)), for now I'm compiling with v4.0 on my Raspi 1B to confirm https://github.com/flightaware/dump1090/issues/115 is fixed in v4.0. Slow machine, but we'll get there at the end of day ;)

Thom-x, 2021-03-23:
fixed by #48

closed 2021-03-23 by Thom-x