Thom-x/docker-fr24feed-piaware-dump1090 #44: Better UI integration with FlightRadar24 in addition to FlightAware

e2jk, 2021-02-28:
## Specifications

  - Image hash or version: [latest/1.3.1](https://hub.docker.com/layers/thomx/fr24feed-piaware/1.3.1/images/sha256-4e09b38fad0db1079e387eccddf4414fc5fc67eb21996ec482be23b02f438ca1?context=explore), contains [v3.8.1 of flightaware/dump1090](https://github.com/flightaware/dump1090/releases/tag/v3.8.1).
  - Platform: Raspberry Pi 1B
  - Subsystem: armv6

## Expected Behavior
As this Docker image combines both PiAware and FR24feed, I'd expect the UI to provide not just links to FlightAware but also to FlightRadar24, and a link to the local FR24Feed page, typically http://localhost:8754/
And why not, also a link to the public pages of one's own user/tracker on both platforms so that you can easily check the health of your radar, as seen externally by those 2 tracking sites:
- https://flightaware.com/adsb/stats/user/e2jk
- https://www.flightradar24.com/account/feed-stats/?id=34513

## Actual Behavior
Data upload is performed to both platforms, which is great.
But the UI elements are only FlightAware-focused
- Logos (which I don't really care about), FlightAware and PiAware/SkyAware
- Page title ("FlightAware SkyView")
- Links to the flights pages, both clicking on the flight number link or the "Visit Flight Page" button when a flight is selected:

[image]

I don't expect FlightAware to modify their version of the dump1090 package to include links to one of their competitors, though ;)
What would be the most efficient way to get these changes, would we need to fork after each stable release and add these items/ Probably better than to try to monkey-patch directly from the Dockerfile...
Any interest to work (together) on this effort? I could create a new repo that downloads the latest stable release of flightaware/dump1090 and modifies/adds these elements, would that be something you'd be interested in?

The links to the user/tracker page on both platforms could be passed on as environment variables when launching the Docker container, like is done for all the other parameters of this container.

**Note:** I am not professionally affiliated to any of the 2 platforms in any way, just a simple user.
I have been a FR24 user for a long time (started my first receiver in 2013, pushing only to FR24) and just recently created my FlightAware profile and pushing to FA thanks to this repo. If it hadn't been for the easy integration with the 2 platforms that is facilitated by your Docker container @Thom-x, I would likely only have continued to push data to FR24 and not FlightAware.
Just adding this in case my suggestion here would be misinterpreted ("all this hard maintenance work done by FlightAware, and this guy just wants to tag along FR24"), it is thanks to this combination that I'm now also feeding FlightAware...

## Steps to Reproduce the Problem

Regular install of the Docker image.

Thom-x, 2021-02-28:
It could be a good idea ! I don't really have time to do that but I would be happy to accept PR for it (and review them).

I'm not a front end expert, but I could help on the Docker part. 👍

e2jk, 2021-02-28:
Yes, time is an issue ;) I'm putting this on my todo list.

e2jk, 2021-02-28:
Hi @Thom-x , I've set up an initial patching setup that adds the FR24 logo and uses a neutral page name. This is how it looks like until now:

[image]

Before I go too far down that road, could you give it a try and tell me if that is in line with what you'd expect?

I've added a bit of explanation here: https://github.com/e2jk/docker-fr24feed-piaware-dump1090/tree/issue-44/patch
Basically I created a .patch file that modifies 4 files from the upstream release, and that file gets plugged into the Dockerfile.

Note: The patch works fine for v3.8.1 (the current target of "latest"), but fails on v4.0 because upstream added extra whitespace in the `public_html/script.js` file, in the `updatePiAwareOrFlightFeeder()` function that gets patched. Simple to update when v4.0 (or likely 4.1) is ready for usage.

e2jk, 2021-02-28:
Just to add a bit more context:
- The "real" work happens [here](https://github.com/e2jk/docker-fr24feed-piaware-dump1090/commit/5bcd6aefbb175914233a4ad8dc22c3e068b013ff) in the Dockerfile, in which:
  - The FR24 logo gets copied 
  - The [patch/flightradar24.patch](https://github.com/e2jk/docker-fr24feed-piaware-dump1090/blob/issue-44/patch/flightradar24.patch) file gets applied that modifies the index page to add the logo and the js files for a neutral page name

Thom-x, 2021-03-01:
Great ! You can do a PR and we will create a new docker tag to test if it's OK.

e2jk, 2021-03-02:
PR created with the code until now. I'm running a Raspi 1B, when creating/publishing the new Docker tag can you make sure to upload also the arm version? Thanks.

e2jk, 2021-03-07:
Thanks for meeting #45. Since there are more changes coming up in #49, @Thom-x can you reopen this issue?

closed 2021-03-16 by Thom-x

e2jk, 2021-03-16:
Great, thanks for merging both PRs!