Thom-x/docker-fr24feed-piaware-dump1090 #52: [UI][Suggestion] Switch from dump1090 to tar1090

e2jk, 2021-03-14:
(removing the default issue template, as this is not really an issue report but a question/suggestion)

Would it be an idea to switch the UI from [dump1090-fa](https://github.com/flightaware/dump1090) to [tar1090](https://github.com/wiedehopf/tar1090)?
As you can see on https://globe.adsbexchange.com/ it does provide quite an improved UI...

(now that I've spent time working on adapting the dump1090 to add the links to FR24 in #44, maybe not the smartest to move right away to another UI, but maybe food for thought?)

Thom-x, 2021-03-15:
There is a new version of the UI on fa https://github.com/flightaware/dump1090/tree/master/public_html_merged
Maybe we should wait a bit to see if it's worth to change.

Personally I don't really care about all fancies options.

Thom-x, 2021-03-15:
Here is the new filters etc

[image]

closed 2021-06-11 by stale[bot]