Thom-x/docker-fr24feed-piaware-dump1090 #56: Supervisor not listening - Container killing itself on startup - Docker on Unraid

shiner66, 2021-04-08:
## Specifications

  - Image hash or version: ecd73f42ddf66091d1c9c8d2544d1e8fd2bd7d93254c2a4bf2498216df3463a0
  - Platform: Unraid
  - Subsystem: Docker

## Expected Behavior

Container actually starting correctly.

## Actual Behavior

Container fails to start with the following log

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-confd-fr24feed: executing...
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO Backend set to env
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO Starting confd
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO Backend source(s) set to
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO /etc/fr24feed.ini has UID 99 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO /etc/fr24feed.ini has GID 100 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO /etc/fr24feed.ini has mode drwxrwxrwx should be -rwxrwxrwx
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO /etc/fr24feed.ini has md5sum d41d8cd98f00b204e9800998ecf8427e should be d720cdd401cb34929e7ccb1122c7cd53
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: INFO Target config /etc/fr24feed.ini out of sync
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: ERROR rename /etc/.fr24feed.ini467061054 /etc/fr24feed.ini: file exists

2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[267]: FATAL rename /etc/.fr24feed.ini467061054 /etc/fr24feed.ini: file exists
[cont-init.d] 01-confd-fr24feed: exited 1.
[cont-init.d] 02-confd-piaware: executing...
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO Backend set to env
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO Starting confd
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO Backend source(s) set to
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO /etc/piaware.conf has UID 99 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO /etc/piaware.conf has GID 100 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO /etc/piaware.conf has mode drwxrwxrwx should be -rwxrwxrwx
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO /etc/piaware.conf has md5sum d41d8cd98f00b204e9800998ecf8427e should be 02b68abf8a0d00b67789aeb454a293ab
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: INFO Target config /etc/piaware.conf out of sync
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: ERROR rename /etc/.piaware.conf045869135 /etc/piaware.conf: file exists

2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[280]: FATAL rename /etc/.piaware.conf045869135 /etc/piaware.conf: file exists
[cont-init.d] 02-confd-piaware: exited 1.
[cont-init.d] 03-confd-html: executing...
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO Backend set to env
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO Starting confd
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO Backend source(s) set to
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO /usr/lib/fr24/public_html/config.js has UID 99 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO /usr/lib/fr24/public_html/config.js has GID 100 should be 0
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO /usr/lib/fr24/public_html/config.js has mode drwxrwxrwx should be -rwxrwxrwx
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO /usr/lib/fr24/public_html/config.js has md5sum d41d8cd98f00b204e9800998ecf8427e should be 8d80f325c4ec83f92dd4f8d19c759f89
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: INFO Target config /usr/lib/fr24/public_html/config.js out of sync
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: ERROR rename /usr/lib/fr24/public_html/.config.js807195435 /usr/lib/fr24/public_html/config.js: file exists

2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[294]: FATAL rename /usr/lib/fr24/public_html/.config.js807195435 /usr/lib/fr24/public_html/config.js: file exists
[cont-init.d] 03-confd-html: exited 1.
[cont-init.d] 04-confd-layers: executing...
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[307]: INFO Backend set to env
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[307]: INFO Starting confd
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[307]: INFO Backend source(s) set to
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[307]: INFO Target config /usr/lib/fr24/public_html/layers.js out of sync
2021-04-08T15:20:30+02:00 96cd3b1a0b77 /opt/confd/bin/confd[307]: INFO Target config /usr/lib/fr24/public_html/layers.js has been updated
[cont-init.d] 04-confd-layers: exited 0.
[cont-init.d] 05-upintheair: executing...
[cont-init.d] 05-upintheair: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Thu Apr 8 15:20:31 2021 CEST dump1090-fa v5.0 starting up.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[main][i]FR24 Feeder/Decoder
[main][i]Version: 1.0.18-5/generic

[main][i]Built on Mar 4 2016 16:26:55 (devel-d11ca48.git/Linux/x86_64)
[main][i]Copyright 2012-2016 Flightradar24 AB
[main][i]http://flightradar24.com
[main][i]DNS mode: LIBC

FR24 Feeder/Decoder usage:
/fr24feed/fr24feed/fr24feed --fr24key=0123456789ABCDEF| Start feeding process manually
/fr24feed/fr24feed/fr24feed --signup | Sign up and configure
/fr24feed/fr24feed/fr24feed --reconfigure | Change configuration
/fr24feed/fr24feed/fr24feed --version | Show binary version number

s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
[cont-finish.d] executing container finish scripts...
Found Rafael Micro R820T tuner
[cont-finish.d] done.
[s6-finish] waiting for services.
rtlsdr: tuner gain set to 49.6 dB
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```
Seems no matter which keys I do use during startup, in the end it kills itself anyway

## Steps to Reproduce the Problem

  1.Pulled thomx/fr24feed-piaware:latest straight from dockerhub
  Configured the keys following github readme = container kills itself
  Not using any extra keys, but just mapping correctly the container paths and ports = container kills itself

Thom-x, 2021-04-08:
What is your docker host paths  mapping ?

shiner66, 2021-04-08:
Hey there, these are the defaults that Unraid does on its own, I have checked them to be accurate (as they always are)

```
--name='fr24feed-piaware'
--net='bridge' 
-e TZ="Europe/Paris" 
-e HOST_OS="Unraid" 
-p '8080:8080/tcp' 
-p '8754:8754/tcp' 
-v '/mnt/user/appdata/fr24feed-piaware/piaware.conf':'/etc/piaware.conf':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/config.js':'/usr/lib/fr24/public_html/config.js':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/fr24feed.ini':'/etc/fr24feed.ini':'rw' 
--device='/dev/bus/usb/001/006' 
'thomx/fr24feed-piaware'
```

Thom-x, 2021-04-08:
You need to either to remove the mount path and add the missing environment variables.

```
-v '/mnt/user/appdata/fr24feed-piaware/piaware.conf':'/etc/piaware.conf':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/config.js':'/usr/lib/fr24/public_html/config.js':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/fr24feed.ini':'/etc/fr24feed.ini':'rw' 
```

or

To create these files (I think they do not exist, and docker is creating a folder instead, that's why the container is failing).
It cannot create the missing config because a folder with the same name already exists.

```
/mnt/user/appdata/fr24feed-piaware/piaware.conf
/mnt/user/appdata/fr24feed-piaware/config.js
/mnt/user/appdata/fr24feed-piaware/fr24feed.ini
```

shiner66, 2021-04-08:
That's such a weird issue, I have never faced a Docker container creating folders instead of files.
I did delete the folders and create those three files, but the container is still having issues with the S6 Supervisor

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-confd-fr24feed: executing...
[cont-init.d] 01-confd-fr24feed: exited 0.
[cont-init.d] 02-confd-piaware: executing...
[cont-init.d] 02-confd-piaware: exited 0.
[cont-init.d] 03-confd-html: executing...
[cont-init.d] 03-confd-html: exited 0.
[cont-init.d] 04-confd-layers: executing...
2021-04-08T15:41:31+02:00 ccfb43f8b7c1 /opt/confd/bin/confd[286]: INFO Backend set to env
2021-04-08T15:41:31+02:00 ccfb43f8b7c1 /opt/confd/bin/confd[286]: INFO Starting confd
2021-04-08T15:41:31+02:00 ccfb43f8b7c1 /opt/confd/bin/confd[286]: INFO Backend source(s) set to
2021-04-08T15:41:31+02:00 ccfb43f8b7c1 /opt/confd/bin/confd[286]: INFO Target config /usr/lib/fr24/public_html/layers.js out of sync
2021-04-08T15:41:31+02:00 ccfb43f8b7c1 /opt/confd/bin/confd[286]: INFO Target config /usr/lib/fr24/public_html/layers.js has been updated
[cont-init.d] 04-confd-layers: exited 0.
[cont-init.d] 05-upintheair: executing...
[cont-init.d] 05-upintheair: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[main][i]FR24 Feeder/Decoder
[main][i]Version: 1.0.18-5/generic

[main][i]Built on Mar 4 2016 16:26:55 (devel-d11ca48.git/Linux/x86_64)
[main][i]Copyright 2012-2016 Flightradar24 AB
[main][i]http://flightradar24.com
[main][i]DNS mode: LIBC

FR24 Feeder/Decoder usage:
/fr24feed/fr24feed/fr24feed --fr24key=0123456789ABCDEF| Start feeding process manually
/fr24feed/fr24feed/fr24feed --signup | Sign up and configure
/fr24feed/fr24feed/fr24feed --reconfigure | Change configuration
/fr24feed/fr24feed/fr24feed --version | Show binary version number

s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
Thu Apr 8 15:41:32 2021 CEST dump1090-fa v5.0 starting up.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
Found Rafael Micro R820T tuner
[s6-finish] sending all processes the TERM signal.
[main][i]FR24 Feeder/Decoder
[main][i]Version: 1.0.18-5/generic

[main][i]Built on Mar 4 2016 16:26:55 (devel-d11ca48.git/Linux/x86_64)
[main][i]Copyright 2012-2016 Flightradar24 AB
[main][i]http://flightradar24.com
[main][i]DNS mode: LIBC

FR24 Feeder/Decoder usage:
/fr24feed/fr24feed/fr24feed --fr24key=0123456789ABCDEF| Start feeding process manually
/fr24feed/fr24feed/fr24feed --signup | Sign up and configure
/fr24feed/fr24feed/fr24feed --reconfigure | Change configuration
/fr24feed/fr24feed/fr24feed --version | Show binary version number

s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
Thu Apr 8 15:41:32 2021 CEST dump1090-fa v5.0 starting up.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
Found Rafael Micro R820T tuner
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

I didn't enter all the required keys as stated in the wiki though, I just did that test as a proof of thought, and while it isn't yelling about already existing files it still fails at `s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening`.

Is it possibile/expected for the container to be failing in such a way when not provided with all the requested keys upon startup?

I did also try to delete the files/folders althogheter, resulting in the container creating those three folders again instead of the files, supposedly

Thom-x, 2021-04-08:
It creates folders because you are mounting them:

```
-v '/mnt/user/appdata/fr24feed-piaware/piaware.conf':'/etc/piaware.conf':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/config.js':'/usr/lib/fr24/public_html/config.js':'rw' 
-v '/mnt/user/appdata/fr24feed-piaware/fr24feed.ini':'/etc/fr24feed.ini':'rw' 
```
This is the expected behaviour for docker.

The normal way to run this image is to only use environement variables without mounting any file.

closed 2021-04-08 by shiner66

Thom-x, 2021-04-08:
> Then it's docker for Unraid that is not behaving itself mounting anything and everything.

I believe so, we needed to mount these files a while ago, the Unraid config must be out of sync now.

shiner66, 2021-04-08:
> > Then it's docker for Unraid that is not behaving itself mounting anything and everything.
> 
> I believe so, we needed to mount these files a while ago, the Unraid config must be out of sync now.

Maybe it would be worthwhile to mention that in the readme?

Thom-x, 2021-04-08:
We don't need to create files for more than one year now. I don't think I will mention that in the README.
Release are made according to https://semver.org/ so it should be OK.

shiner66, 2021-04-08:
I see what's happened, I picked it up from CA, so I guess I should post on the [maintainer 3D](https://forums.unraid.net/topic/79394-support-foxxmd-fr24feed-piaware). 
Thanks for the help anyway! Nice work on this container, it's working beautifully.
Now I just have to debug that `[mlat][i]No ADS-B time reference available (0/0)`

Thom-x, 2021-04-08:
Thanks !
I got the same log
```
2021-04-08 14:17:52 | [mlat][i]No ADS-B time reference available (0/0)
```
But still MLAT is reported working on flightaware.

shiner66, 2021-04-08:
Hi again!

Can confirm. MLAT needs to see also ADS-B traffic so it can sync up to their time.

```
2021-04-08 22:30:40 | [mlat][i]Received ADS-B time references AC:
2021-04-08 22:30:40 | [mlat][i] 4CA98A
2021-04-08 22:30:40 | [mlat][i] 4D2243
```