Thom-x/docker-fr24feed-piaware-dump1090 #63: Upgrade to V6.0 problem

DanielMcAssey, 2021-09-09:
Issue with new update, getting the following error when launching. It was working before the upgrade.

`warning: /etc/piaware.conf: failed to read config file: error reading "file3": illegal operation on a directory

piaware (process 361) is shutting down because it received a shutdown signal (SIGTERM) from the system...
piaware (process 361) is exiting...
[s6-finish] sending all processes the KILL signal and exiting.
`

Thom-x, 2021-09-09:
Hello,

Are you mounting a volume with `/etc/piaware.conf `? It appears to be a folder. If so, don't mount any volume.

Thom-x, 2021-09-09:
See https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/56#issuecomment-815842310

DanielMcAssey, 2021-09-09:
Spot on. That solved it. Thank you

closed 2021-09-09 by DanielMcAssey