Thom-x/docker-fr24feed-piaware-dump1090 #66: Unable to update to v1.8.0

yayitazale, 2022-02-10:
I just pulled the new image and the docker is unable to start. Do I need to rejoin to FR24 and FlightAware again or what I'm doing wrong?

I'm using a RPI 3b+ and a standard RTL2832U, everything is working good with 1.7.0

The launch command:
```
sudo docker run -d -p 8080:8080 -p 8754:8754 --name rtl --restart=always --device=/dev/bus/usb:/dev/bus/usb -e "FR24FEED_FR24KEY=XXXXXXXX" -e "PIAWARE_FEEDE
R_DASH_ID=XXXXXX" -e "HTML_SITE_LAT=XXXXX" -e "HTML_SITE_LON=XXXX" -e "HTML_SITE_NAME=Lasarte" -e "PANORAMA_ID=XXXXXX" thomx/fr24feed-piaware
```

The logs:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-confd-fr24feed: executing... 
/var/run/s6/etc/cont-init.d/01-confd-fr24feed: line 23: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 01-confd-fr24feed: exited 127.
[cont-init.d] 02-confd-piaware: executing... 
/var/run/s6/etc/cont-init.d/02-confd-piaware: line 18: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 02-confd-piaware: exited 127.
[cont-init.d] 03-confd-html: executing... 
/var/run/s6/etc/cont-init.d/03-confd-html: line 21: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 03-confd-html: exited 127.
[cont-init.d] 04-confd-layers: executing... 
/var/run/s6/etc/cont-init.d/04-confd-layers: line 3: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 04-confd-layers: exited 127.
[cont-init.d] 05-upintheair: executing... 
--2022-02-10 08:04:13--  http://www.heywhatsthat.com/api/upintheair.json?id=M93W7NC1&refraction=0.25&alts=1000,10000
Resolving www.heywhatsthat.com (www.heywhatsthat.com)... 69.39.100.105
Connecting to www.heywhatsthat.com (www.heywhatsthat.com)|69.39.100.105|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/json]
Saving to: '/usr/lib/fr24/public_html/upintheair.json'

     0K .......... .......... ......                            152K=0.2s

2022-02-10 08:04:14 (152 KB/s) - '/usr/lib/fr24/public_html/upintheair.json' saved [26641]

[cont-init.d] 05-upintheair: exited 0.
[cont-init.d] done.
[services.d] starting services
Thu Feb 10 08:04:14 2022 UTC  dump1090-fa v7.1 starting up.
libcommon read_config failed - /etc/fr24feed.ini
______  _  _         _      _                    _              _____    ___ 
|  ___|| |(_)       | |    | |                  | |            / __  \  /   |
| |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
|  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
| |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
\_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
               __/ |                                                         
              |___/                                                          
[main][i]FR24 Feeder/Decoder
[main][i]Version: 1.0.25-3/generic
[main][i]Built on Apr  8 2020 07:06:55 (HEAD-5e8ef0c.git/Linux/static_armel)
[main][i]Running on: debian10
[main][i]Local IP(s): 172.17.0.2
[main][i]Copyright 2012-2020 Flightradar24 AB
[main][i]https://www.flightradar24.com
[main][i]DNS mode: PING

FR24 Feeder/Decoder usage:
/fr24feed/fr24feed/fr24feed --fr24key=0123456789ABCDEF| Start feeding process manually
/fr24feed/fr24feed/fr24feed --signup                  | Sign up and configure
/fr24feed/fr24feed/fr24feed --reconfigure             | Change configuration
/fr24feed/fr24feed/fr24feed --version                 | Show binary version number
[services.d] done.
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
s6-svscanctl: fatal: unable to control /var/run/s6/services: supervisor not listening
[cont-finish.d] executing container finish scripts...
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

Thanks

Thom-x, 2022-02-10:
Hi, 

Fr24feed ini file is not good => `libcommon read_config failed - /etc/fr24feed.ini`
You could try to start the container again, but this time overriding the ini script for fr24 ?
Or you can try to disable fr24feed to see if the container starts.

On my side I have the exact same config, no problem with fr24feed.
Be sure that you are not mounting any (bad) volume into the container.

mount `/etc/cont-init.d/01-confd-fr24feed` with this content (`cat` the ini file to see if it is good).

```
#!/usr/bin/with-contenv bash

default_value() 
{ 
        key=${1//\-/_DASH_}
        key=FR24FEED_${key^^}
        eval "value=\${$key:-\$2}"
        printf -v $key -- "$value"
        export $key
}

default_value "receiver" "beast-tcp"
default_value "fr24key" "YOUR_KEY_HERE"
default_value "host" "127.0.0.1:30005"
default_value "bs" "no"
default_value "raw" "no"
default_value "logmode" "1"
default_value "logpath" "/tmp"
default_value "mlat" "yes"
default_value "mlat-without-gps" "yes"

if [ ! -f /etc/fr24feed.ini ]; then
        /opt/confd/bin/confd -onetime -confdir "/etc/confd/fr24feed" -prefix "fr24feed" -backend env
       cat /etc/fr24feed.ini
else
        echo "init file already existing"
        cat /etc/fr24feed.ini
fi
```

yayitazale, 2022-02-10:
With FR24 disabled the container starts but I can't see the map:

https://fr.yayitazale.cc/

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-confd-fr24feed: executing... 
/var/run/s6/etc/cont-init.d/01-confd-fr24feed: line 23: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 01-confd-fr24feed: exited 127.
[cont-init.d] 02-confd-piaware: executing... 
/var/run/s6/etc/cont-init.d/02-confd-piaware: line 18: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 02-confd-piaware: exited 127.
[cont-init.d] 03-confd-html: executing... 
/var/run/s6/etc/cont-init.d/03-confd-html: line 21: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 03-confd-html: exited 127.
[cont-init.d] 04-confd-layers: executing... 
/var/run/s6/etc/cont-init.d/04-confd-layers: line 3: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 04-confd-layers: exited 127.
[cont-init.d] 05-upintheair: executing... 
--2022-02-10 09:06:39--  http://www.heywhatsthat.com/api/upintheair.json?id=M93W7NC1&refraction=0.25&alts=1000,10000
Resolving www.heywhatsthat.com (www.heywhatsthat.com)... 69.39.100.105
Connecting to www.heywhatsthat.com (www.heywhatsthat.com)|69.39.100.105|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/json]
Saving to: '/usr/lib/fr24/public_html/upintheair.json'

     0K .......... .......... ......                            155K=0.2s

2022-02-10 09:06:40 (155 KB/s) - '/usr/lib/fr24/public_html/upintheair.json' saved [26641]

[cont-init.d] 05-upintheair: exited 0.
[cont-init.d] done.
[services.d] starting services
Thu Feb 10 09:06:40 2022 UTC  dump1090-fa v7.1 starting up.
[services.d] done.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
Found Rafael Micro R820T tuner
****************************************************
piaware version 7.1 is running, process ID 337
your system info is: Linux 46892834b92a 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux
rtlsdr: tuner gain set to 49.6 dB (gain step 28)
Allocating 4 zero-copy buffers
Connecting to FlightAware adept server at piaware.flightaware.com/1200
Connection with adept server at piaware.flightaware.com/1200 established
TLS handshake with adept server at piaware.flightaware.com/1200 completed
FlightAware server certificate validated
encrypted session established with FlightAware
logged in to FlightAware as user guest
my feeder ID is 9c825fd0-65b5-4df6-931c-ef8f3293eba0
ADS-B data program 'dump1090' is listening on port 30005, so far so good
Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30005 --stdout
Started faup1090 (pid 365) to connect to dump1090
UAT support disabled by local configuration setting: uat-receiver-type
piaware received a message from dump1090!
piaware has successfully sent several msgs to FlightAware!
11 msgs recv'd from dump1090; 11 msgs sent to FlightAware
```

Thom-x, 2022-02-10:
I think confd is not working well with  RPI 3b+ ...
That's why it is failing. Could you try to shell into the container and do `/opt/confd/bin/confd` ?

```
/var/run/s6/etc/cont-init.d/01-confd-fr24feed: line 23: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 01-confd-fr24feed: exited 127.
[cont-init.d] 02-confd-piaware: executing... 
/var/run/s6/etc/cont-init.d/02-confd-piaware: line 18: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 02-confd-piaware: exited 127.
[cont-init.d] 03-confd-html: executing... 
/var/run/s6/etc/cont-init.d/03-confd-html: line 21: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 03-confd-html: exited 127.
[cont-init.d] 04-confd-layers: executing... 
/var/run/s6/etc/cont-init.d/04-confd-layers: line 3: /opt/confd/bin/confd: No such file or directory
[cont-init.d] 04-confd-layers: exited 127.
[cont-init.d] 05-upintheair: executing... 
```

yayitazale, 2022-02-10:
```
root@46892834b92a:~# /opt/confd/bin/confd
bash: /opt/confd/bin/confd: No such file or directory
root@46892834b92a:~# cd /opt/confd/bin/      
root@46892834b92a:/opt/confd/bin# ls
confd
```

Thom-x, 2022-02-10:
Ok the bin is not working for your distro.
Could you extract confd bin from the previous working image and upload it here (I don't have PI to do it myself)?
I will put it in place of this non-working one.

Also, could you give me the output of `dpkg --print-architecture` on RPI 3b+?

yayitazale, 2022-02-10:
Sorry is not a B+ is just a Raspberry Pi 3 Model B

```
# dpkg --print-architecture
armhf
```

Not sure how to extract the confd bin with image 1.7.0...

Now with 1.7.0 the map is not working either...

Thom-x, 2022-02-10:
>Now with 1.7.0 the map is not working either...

This is very strange, Nothing should be kept from the previous container.
Try to look at the browser console, but if you have messages about:

[image]

It's confd not working.

To extract the bin from the container, you can first try the same command as before `/opt/confd/bin/confd` to be sure it is working. And from your host:

```
docker cp containerId:/opt/confd/bin/confd /path/on/host/confd
```

yayitazale, 2022-02-10:
Is weird, now is working, I think maybe is something related with the caches of internet.

There you have the confd file

[confd.zip](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/files/8040340/confd.zip)

Thom-x, 2022-02-10:
Could you try this image version ?
https://hub.docker.com/layers/191749382/thomx/fr24feed-piaware/PR-67/images/sha256-eca9e2888f992948460fce4269140c689a09f83046ad6bb73ae9f8eba09fad77?context=repo

PR #67 (2022-02-10): fix(confd): replace armhf condf bin with working one
Fix #66 corrupted confd bin 

```
bash: /opt/confd/bin/confd: No such file or directory
```