Thom-x/docker-fr24feed-piaware-dump1090 #70: Add adsbexchange feed client?

wiedehopf, 2022-02-20:
I know the name is very specific but i don't think it would preclude adding another feed client as an option to the users.
(and obviously i'm not asking for a change in name ;) )

If that would be an option for you, i'd work on it and make a pull request eventually.
I wouldn't expect to add more than 20 MB of size to the image even when including an extra interface (tar1090 using data from the feed client).
The whole code wouldn't run unless people define ADSBEXCHANGE_UUID environment variable or something along those lines.

Let me know!

Thom-x, 2022-02-20:
Hi, it would be nice !
We could follow the same pattern as for fr24 and fa, a global flag to enable it SERVICE_ENABLE_ADSBEXCHANGE, and extra parameters vars needed.

Delta1977, 2022-03-05:
Hi, this feature would be nice

Thom-x, 2022-03-17:
fixed by #80
Thank you very much @wiedehopf !