Thom-x/docker-fr24feed-piaware-dump1090 #77: openweathermap.org not working with free version

kalazzerx, 2022-03-04:
I was trying to use the OWM and it was not working after I added the API key (free version).  I did some poking around and it appears that the map API calls used are not supported in the free version.   From what I could tell the 'advanced weather maps' API is currently being called which requests a paid subscription (https://openweathermap.org/price#weather).

These are the errors I am seeing in my browser console

`GET https://a.sat.owm.io/vane/2.0/weather/WS10/8/67/102?appid=<myapikey was here> 401 (Unauthorized)`

page content: 

`{"cod":401, "message": "Invalid API key. Please see http://openweathermap.org/faq#error401 for more info."}`

401 is an unauthorized error per the faq page - I did run an API call against basic maps with my API key and did not have an issue.

Advance weather maps API call example
`http://maps.openweathermap.org/maps/2.0/weather/{op}/{z}/{x}/{y}?appid=[{API key}`

Basic weather maps API call example
`https://tile.openweathermap.org/map/{layer}/{z}/{x}/{y}.png?appid={API key}`

If that is a known requirement then would be nice to add that to the documentation.  If I am totally missing something then let me know.

BTW, thank you for the great work building this!  I appreciate what you are doing.

Thom-x, 2022-03-04:
Hi, it was probably free before and it is not now. I'll update the tiles to use the free weather URL 👍.

Thom-x, 2022-03-07:
I updated the doc, you need a paid account. My key is working for a bit of time, but I run out of free requests after some time.

closed 2022-03-07 by Thom-x

shadowpanther, 2024-07-03:
Hey @Thom-x is there a possibility to use Weather maps 1.0 API (available for a Free subscription tier) instead of 2.0 (available for 140GBP/mo)?
1.0 API description is here: https://openweathermap.org/api/weathermaps

Looks like x/y/z is the same, but there are fewer layers and layer names are different in 1.0 compared to 2.0.