Thom-x/docker-fr24feed-piaware-dump1090 #95: Disable mlat for Flightradar24

jkldgoefgkljefogeg, 2022-06-06:
FR24 sent out an email asking feeders to disable MLAT

> Important information
> If you intend to share data to networks alongside Flightradar24, in your Flightradar24 receiver please disable MLAT to the following settings: MLAT=“no”and MLAT-without-gps=“no”. This is to ensure the quality of the data we receive and use and to reduce incompatibility with other services. 

FR24FEED_MLAT and FR24FEED_MLAT_DASH_WITHOUT_DASH_GPS should default to False?

Thom-x, 2022-06-09:
I read on the forum people saying they should rather filter on their side.
I kind of agree with them, why having this option if we mustn't activate it ?🤷‍♂️

ignisf, 2022-09-29:
How would they be able to filter it on their site though? I can imagine the issue -- they may be ingesting data from other services on their side, and you are feeding both those services and FR24 directly. How would they identify data from your system coming through other services so that they can filter it out?

Additionally:

1. They asked nicely. After all you're voluntarily feeding them data, doesn't it make sense to stop feeding them part of the data when they told you it's giving them trouble?
2. > why having this option if we mustn't activate it
   
   So one can activate it in case they're feeding it only to fr24.

Anyway, it's configurable, so whatever

MaxWinterstein, 2022-11-25:
It's now disabled per default - see #98

closed 2022-11-25 by Thom-x