Thom-x/docker-fr24feed-piaware-dump1090 #99: [Feature request] Replace dump1090-fa with tar1090

serniko97, 2022-11-01:
Yesterday I noticed that the map on ADS-B Exchange had a similar but much better UI than the one available here on port 8080.
Therefore, since dump1090-fa which lacks many features and doesn't support mobile browsers, it would be a very nice improvement using [tar1090](https://github.com/wiedehopf/tar1090) instead as it provides a much better UI and UX.

Is there any downside to this?

Thom-x, 2022-11-01:
Hi, you can easily connect this image with tar-1090, that's actually what I'm doing. I don't plan to add it to the image, as it focus on sharing data to different providers and not on visualizing.

serniko97, 2022-11-04:
Hi @Thom-x, which configurations are you using to feed tar1090? I have some trouble with the meaning of certain parameters.
Both tar1090 and this image will be running on the same server.

Thom-x, 2022-11-08:
```
version: "3"
services:
  tar1090:
    image: ghcr.io/sdr-enthusiasts/docker-tar1090:latest
    restart: unless-stopped
    ports:
      - "80:80"
    environment:
        - "TZ=Europe/Paris"
        - "BEASTHOST=fr24feed-piaware" # <--- fr24 feed container hostname
        - "LAT=xxx"
        - "LONG=xxx"
        - "MLATHOST=fr24feed-piaware" # <--- fr24 feed container hostname
        - "ENABLE_TIMELAPSE1090=true"
    volumes:
      - "xxx:/var/globe_history"
      - "xxx:/var/timelapse1090"
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```

closed 2022-11-25 by Thom-x