Thom-x/docker-fr24feed-piaware-dump1090 #103: option to select USB device

Rothammel, 2023-01-12:
I have two rtl devices, one for my 433MHz temperature sensors, and one for ADS-B Multi-Portal Feeder

Thom-x, 2023-01-12:
You can select the one you want to share with replacing `--device=/dev/bus/usb:/dev/bus/usb \` with the exact usb port you want to use.

Rothammel, 2023-01-12:
thx, but I was wrong, I use the Home Assistant addon from Max

Thom-x, 2023-01-12:
In this case you should ask him to do that. I cannot change this parameter from here.

martian, 2023-01-20:
You can set the serial numbers in the devices using rtl_eeprom (if your dongle has an eeprom). Those serials can be used to select the correct device in dump1090 and rtl_433. This also works in the Home Assistant addons.

I use serials 00000433 and 00001090.
For dump1090, set `DUMP1090_ADDITIONAL_ARGS = -device 00001090`
For rtl_433, edit rtl_433.conf[.template] and use:  `device :00000443`

MaxWinterstein, 2023-01-30:
Owner of the add-on, aka Max here:

Providing dump1090 args can be archived using this field:

[image]

For reference: https://github.com/MaxWinterstein/homeassistant-addons/issues/134

Rothammel, 2023-01-30:
Awesome, it works. thank you all so much

[image]
[image]
[image]

closed 2023-01-30 by Rothammel