Thom-x/docker-fr24feed-piaware-dump1090 #104: Update fr24 version to 1.0.34-0

bpoetzschke, 2023-01-19:
Hi there,

I just received and email from fr24 stating that version 1.0.25-3 will no longer be supported after Feb 28 2023.
I updated to the most recent image 1.18.0 but it still runs on the 1.0.25-3 version of fr24.

Thom-x, 2023-01-19:
Yep for rpi there is a version but not other arch.

maweki, 2023-01-19:
> Yep for rpi there is a version but not other arch.

So who writes to their support address?

Thom-x, 2023-01-19:
Let's all ask them, we will have more chance to have a reply 😄

Thom-x, 2023-01-19:
Download page has been updated also 🥳

kalazzerx, 2023-01-19:
Also got the warning about the version today.  Same message as posted above.

Littlejth, 2023-01-19:
Just pulled it down on my Pi Zero 2 W running Manjaro ARM and doesn't appear to be any issues at the present time. Will report if there are any but otherwise so far so good, thanks for the quick work on this!

yayitazale, 2023-01-19:
Running with no issues with a RPI b3+. Thanks!

bpoetzschke, 2023-01-19:
Running on my RPI 4 without any issues as well. Thank you for the quick fix.

mwgg, 2023-01-20:
Working with no issues on amd64 Ubuntu 20.04. Thanks!

marcelloceschia, 2023-01-20:
RPI@Ubuntu 22.04.1 LTS works well too, thank you for this quick change :)

closed 2023-01-20 by Thom-x

Thom-x, 2023-01-20:
Is now available with latest or 1.19.0 👍

jakubmuransky, 2023-02-13:
Hello,
I'm getting error 
`Setting up fr24feed (1.0.34-0) ...
useradd: user 'fr24' already exists
chmod: missing operand after ‘u+s’
Try 'chmod --help' for more information.
dpkg: error processing package fr24feed (--configure):
 installed fr24feed package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 fr24feed
E: Sub-process /usr/bin/dpkg returned an error code (1)
Failed to update, restarting current version...`
Same with the manual install....