Thom-x/docker-fr24feed-piaware-dump1090 #107: Container crashing without any apparent reason

fl0wfr, 2023-01-21:
## Specifications

  - Image hash or version: PR-106 and latest
  - Platform: amd64
  - Subsystem: 

## Expected Behavior
Simply start and feed FR24/PiAware

## Actual Behavior
Crash loop without any error

## Steps to Reproduce the Problem

  1. Start the container with this command: docker run -d -p 8080:8080 -p 8754:8754 \
	--device=/dev/bus/usb:/dev/bus/usb \
	-v "/etc/localtime:/etc/localtime:ro" \
	-e "FR24FEED_FR24KEY=<>" \
	-e "PIAWARE_FEEDER_DASH_ID=<>" \
	-e "HTML_SITE_LAT=<>" \
	-e "HTML_SITE_LON=<>" \
	-e "HTML_SITE_NAME=<>" \
	-e "SERVICE_ENABLE_ADSBEXCHANGE=false" \
	-e "MLAT_EXACT_LAT=<>" \
	-e "MLAT_EXACT_LON=<>" \
	-e "MLAT_ALTITUDE_MSL_METERS=<>" \
	-e "SERVICE_ENABLE_PLANEFINDER=false" \
	-e "TZ=Etc/GMT+0" \
	--tmpfs /run:exec,size=32M \
	--tmpfs /planefinder/log:exec,size=32M \
	--tmpfs /usr/lib/fr24/public_html/data:size=32M \
	--restart=unless-stopped \
	--network=myNetwork \
	--name fr24feed \
	thomx/fr24feed-piaware:PR-106
  2. USB Tuner is: Bus 002 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

Some logs:
`s6-rc: info: service adsbexchange-feed: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service conf-planefinder successfully started
s6-rc: info: service planefinder: starting
s6-rc: info: service confd-fr24feed successfully started
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[96]: INFO Backend set to env
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[96]: INFO Starting confd
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[96]: INFO Backend source(s) set to 
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[96]: INFO Target config /etc/piaware.conf out of sync
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[96]: INFO Target config /etc/piaware.conf has been updated
s6-rc: info: service fr24feed: starting
s6-rc: info: service adsbexchange-feed successfully started
s6-rc: info: service adsbexchange-stats: starting
[dump1090] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2832UHIDIR, SN 00000001)
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[112]: INFO Backend set to env
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[112]: INFO Starting confd
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[112]: INFO Backend source(s) set to 
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[111]: INFO Backend set to env
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[111]: INFO Starting confd
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[111]: INFO Backend source(s) set to 
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[111]: INFO Target config /usr/lib/fr24/public_html/config.js out of sync
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[111]: INFO Target config /usr/lib/fr24/public_html/config.js has been updated
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[112]: INFO Target config /usr/lib/fr24/public_html/layers.js out of sync
2023-01-21T15:56:19Z 88b2c8ae6db7 /opt/confd/bin/confd[112]: INFO Target config /usr/lib/fr24/public_html/layers.js has been updated
s6-rc: info: service adsbexchange-mlat: starting
s6-rc: info: service confd-piaware successfully started
s6-rc: info: service piaware: starting
s6-rc: info: service confd-html successfully started
s6-rc: info: service planefinder successfully started
s6-rc: info: service planefinder-log: starting
s6-rc: info: service fr24feed successfully started
s6-rc: info: service adsbexchange-stats successfully started
s6-rc: info: service adsbexchange-mlat successfully started
s6-rc: info: service confd-layers successfully started
s6-rc: info: service http: starting
s6-rc: info: service piaware successfully started
s6-rc: info: service planefinder-log successfully started
s6-rc: info: service http successfully started
s6-rc: info: service legacy-services: starting
[fr24feed] 2023-01-21 15:56:19 | ______  _  _         _      _                    _              _____    ___ 
[fr24feed] 2023-01-21 15:56:19 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |
[fr24feed] 2023-01-21 15:56:19 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
[fr24feed] 2023-01-21 15:56:19 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
[fr24feed] 2023-01-21 15:56:19 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
[fr24feed] 2023-01-21 15:56:19 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
[fr24feed] 2023-01-21 15:56:19 |                __/ |                                                         
[fr24feed] 2023-01-21 15:56:19 |               |___/                                                          
[fr24feed] 2023-01-21 15:56:19 | [main][i]FR24 Feeder/Decoder
[fr24feed] 2023-01-21 15:56:19 | [main][i]Version: 1.0.34-0/generic
[fr24feed] 2023-01-21 15:56:19 | [main][i]Built on Jan 10 2023 10:09:01 (decoder-v1.0.34-0-7c5301b.git/Linux/static_amd64)
[fr24feed] 2023-01-21 15:56:19 | [main][i]Running on: debian="11"
[fr24feed] 2023-01-21 15:56:19 | [main][i]Local IP(s): 172.19.0.6
[fr24feed] 2023-01-21 15:56:19 | [main][i]Copyright 2012-2023 Flightradar24 AB
[fr24feed] 2023-01-21 15:56:19 | [main][i]https://www.flightradar24.com
[fr24feed] 2023-01-21 15:56:19 | [main][i]DNS mode: PING
[fr24feed] 2023-01-21 15:56:19 | [main][i]Automatic updates are DISABLED
[fr24feed] 2023-01-21 15:56:19 | 23-01-21 15:56:19.309 [I][http-server.cpp:270] [httpd]Server started, listening on :::8754
s6-rc: info: service legacy-services successfully started
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service planefinder-log: stopping
s6-rc: info: service piaware: stopping
s6-rc: info: service netcat: stopping
s6-rc: info: service http: stopping
s6-rc: info: service fr24feed: stopping
s6-rc: info: service adsbexchange-stats: stopping
s6-rc: info: service adsbexchange-mlat: stopping
s6-rc: info: service http successfully stopped
s6-rc: info: service init-upintheair: stopping
s6-rc: info: service confd-layers: stopping
s6-rc: info: service confd-html: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service planefinder-log successfully stopped
s6-rc: info: service planefinder: stopping
s6-rc: info: service piaware successfully stopped
s6-rc: info: service confd-piaware: stopping
s6-rc: info: service netcat successfully stopped
s6-rc: info: service fr24feed successfully stopped
s6-rc: info: service confd-fr24feed: stopping
s6-rc: info: service adsbexchange-stats successfully stopped
s6-rc: info: service init-upintheair successfully stopped
s6-rc: info: service confd-layers successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service confd-html successfully stopped
s6-rc: info: service adsbexchange-mlat successfully stopped
s6-rc: info: service adsbexchange-feed: stopping
s6-rc: info: service planefinder successfully stopped
s6-rc: info: service conf-planefinder: stopping
s6-rc: info: service confd-fr24feed successfully stopped
s6-rc: info: service confd-piaware successfully stopped
s6-rc: info: service conf-planefinder successfully stopped
s6-rc: info: service adsbexchange-feed successfully stopped
s6-rc: info: service conf-adsbexchange: stopping
s6-rc: info: service dump1090: stopping
s6-rc: info: service conf-adsbexchange successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service dump1090 successfully stopped`

Thom-x, 2023-01-23:
Could you try disabling services one by one to see the culprit ?

Thom-x, 2023-01-23:
Ok, at least everything else is OK.
Could you try to start manually the HTTP server by executing the same command ? 
https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/1afb9b3f971801f905abb9d70b036a290013326a/root/etc/s6-overlay/s6-rc.d/http/run#L5

fl0wfr, 2023-01-23:
No return of the command, I don't have ps in the image so I can't check if it's running, but I doubt it runs. Also, no message in logs following the start of thttpd. And no access to the page with the published port.

Thom-x, 2023-01-23:
Probably not working (it should be blocking). You coucld check the exit code `$?`.
Could you change the port by a random one `-p 4242`?

fl0wfr, 2023-01-23:
Same issue with port 4242, and I get a "1" exit code.

Thom-x, 2023-01-23:
Strange, I would suggest changing path, ports, etc to see what's wrong.
I'm not sure if I can help more. 😕

fl0wfr, 2023-01-23:
Okay, seems strange, I just popped the container on a new host, no persistent storage, basic options. In the meantime I will run without the WebUI.
Thanks anyway!

Thom-x, 2023-01-23:
Sorry for that :/

fl0wfr, 2023-01-23:
No worries, you already did a great work, thanks!

rpavlyuk, 2023-02-09:
I can confirm the same problem: as soon as you enable SERVICE_ENABLE_HTTP -- it crashes. Starting `thttpd` inside the container crashes it as well.

@Thom-x, if you can provide any instructions on how and where to check `thttpd` logs (I'm RedHat guy, not Debian :)) so I can troubleshot the issue.

Thom-x, 2023-02-14:
>@Thom-x, if you can provide any instructions on how and where to check thttpd logs (I'm RedHat guy, not Debian :)) so I can troubleshot the issue.

`thttpd `is logging on stdout (no daemon), what strange is that I'm also using it on amd64 without error.

Thom-x, 2023-03-02:
@Littlejth Did you try starting the container with only HTTPD enabled ?

Littlejth, 2023-03-02:
With only the HTTP service enabled, it still just starts and stops again. Attached is the log file of a single time I attempted to start it after creating the container with only HTTP enabled.

[fr24feed-piaware.log](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/files/10876407/fr24feed-piaware.log)

Thom-x, 2023-03-03:
I created a branch to switch from thttpd to httpd, I'll let you now when the image is available so you can test and give me the feedback https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/pull/114.

Thom-x, 2023-03-04:
You can test it with `docker pull thomx/fr24feed-piaware:PR-114`

fl0wfr, 2023-03-04:
Hello Thom-x,
Same issue, and I see this message in logs:
`[http] /httpd-foreground: 7: exec: httpd: not found`

Thom-x, 2023-03-04:
What is the content of `/usr/local/apache2/bin` it's probably not the same between different arch ?

fl0wfr, 2023-03-04:
[image]

Thom-x, 2023-03-04:
Ok funny, it's probably not compatible for all arch...
I need to find another candidate 😄.

Littlejth, 2023-03-08:
It did start working on my aarch64 pi but obviously not a great solution if it doesn't work for AMD64 as well :)

MaxWinterstein, 2023-04-20:
@fl0wfr which docker version are you running? v23?

Users of my _Home Assistant_ Add-On raised the same issue: https://github.com/MaxWinterstein/homeassistant-addons/issues/149

I guess, `thttpd` does what is explained here: https://developers.home-assistant.io/blog/2023/04/13/new_limits_for_add_ons/

>With Home Assistant OS 10, we update to the latest Docker release 23.0. With the new Docker version the maximum number of open file descriptors for add-ons changed to infinity (sort of).
> ...
> Also in all three cases, memory got allocated dynamically depending on the number of open file descriptors allowed (which can be determined via prlimit64 syscall, which returns a limit of 1073741816).

fl0wfr, 2023-04-20:
@MaxWinterstein yes I'm using Docker 23.0.4

MaxWinterstein, 2023-04-20:
@fl0wfr no linux memory expert here, but can you give the `ulimit` idea a try?

https://docs.docker.com/engine/reference/commandline/run/#ulimit

Something like
```bash
docker run --ulimit nofile= 1048576 ......
```

might be worth a shot

fl0wfr, 2023-04-20:
@MaxWinterstein it works now. That should do the trick for now, thanks.

MaxWinterstein, 2023-04-20:
@Thom-x this might hit more and more people who switch to docker 23

I created #118 as quickfix, but switching to some more recent webserver might be a better approach ;)

Thom-x, 2023-04-20:
Thanks, I'll do a PR image for testing quickly and release of it's all good 👍 !

MaxWinterstein, 2023-04-20:
> Thanks, I'll do a PR image for testing quickly and release of it's all good 👍 !

Awesome! Can't really test it myself here, hope it works 🤞

fl0wfr, 2023-04-20:
You let me know and I will test it ;)

CoMPaTech, 2023-04-20:
I still ran into issues with Max' update on his HA-addon. Still OOM errors on thttp. See screenshot in https://github.com/MaxWinterstein/homeassistant-addons/issues/149#issuecomment-1516359955 if helpful

MaxWinterstein, 2023-04-20:
> I still ran into issues with Max' update on his HA-addon. Still OOM errors on thttp. See screenshot in [MaxWinterstein/homeassistant-addons#149 (comment)](https://github.com/MaxWinterstein/homeassistant-addons/issues/149#issuecomment-1516359955) if helpful

I guess the fix in my addon is placed wrongly. I think s6 starts the httpd stuff in another shell/session/thread/... where my adjustment is not read.

#118 is made a little bit different

Thom-x, 2023-04-25:
You can try with the image `thomx/fr24feed-piaware:PR-118`

fl0wfr, 2023-04-26:
It works, thank you :)

Thom-x, 2023-04-26:
Nice 😄 let's do a release then.

Thom-x, 2023-04-26:
It's now available with 1.21.0 or latest 👍

closed 2023-04-26 by Thom-x

TheDom42, 2023-04-27:
> It's now available with 1.21.0 or latest 👍

Running Unraid 6.11.5 (latest stable version), which has Docker v20.10.21.

At least on my system, this update to 1.21.0 does not seem to work. After I update, my (otherwise so far completely stable) container stops automatically without any error logged. This happens a few seconds after start of the container. 

Downgrading back to v1.20.0 fixes the issue, so I would assume, that this is related to this issue? If I have time later on, I can try the dedicated PR tag/image, if that gives a clearer picture.

fl0wfr, 2023-04-27:
> > It works, thank you :)
> 
> @fl0wfr did you see any difference with the cpu usage?

No, around 3 and 7% CPU usage, running on a Celeron N5105.