Thom-x/docker-fr24feed-piaware-dump1090 #109: Doker on UNRAID refuises to run

miodzicho, 2023-01-27:
## Specifications

  - Image hash or version: latest
  - Platform: UNRAID
  - Subsystem:

## Expected Behavior
docker should start

## Actual Behavior
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: INFO /etc/piaware.conf has mode drwxr-xr-x should be -rwxr-xr-x
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: INFO /etc/piaware.conf has md5sum d41d8cd98f00b204e9800998ecf8427e should be c17037a2b98e96c8a5cb3ec7794b82ea
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: INFO Target config /etc/piaware.conf out of sync
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: ERROR rename /etc/.piaware.conf636670703 /etc/piaware.conf: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: FATAL rename /etc/.piaware.conf636670703 /etc/piaware.conf: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: INFO /etc/fr24feed.ini has mode drwxr-xr-x should be -rwxr-xr-x
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: INFO /etc/fr24feed.ini has md5sum d41d8cd98f00b204e9800998ecf8427e should be 36c55057f9f68fef63c39a0860858718
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: INFO Target config /etc/fr24feed.ini out of sync
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: ERROR rename /etc/.fr24feed.ini800013444 /etc/fr24feed.ini: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: FATAL rename /etc/.fr24feed.ini800013444 /etc/fr24feed.ini: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: INFO /usr/lib/fr24/public_html/config.js has mode drwxr-xr-x should be -rwxr-xr-x
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: INFO /usr/lib/fr24/public_html/config.js has md5sum d41d8cd98f00b204e9800998ecf8427e should be 8538a54a6ad0763a4aad43aba067cecd
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: INFO Target config /usr/lib/fr24/public_html/config.js out of sync
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: ERROR rename /usr/lib/fr24/public_html/.config.js963024346 /usr/lib/fr24/public_html/config.js: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: FATAL rename /usr/lib/fr24/public_html/.config.js963024346 /usr/lib/fr24/public_html/config.js: file exists
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[96]: INFO Backend set to env
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[96]: INFO Starting confd
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[96]: INFO Backend source(s) set to 
s6-rc: warning: unable to start service confd-html: command exited 1
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[96]: INFO Target config /usr/lib/fr24/public_html/layers.js out of sync
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[96]: INFO Target config /usr/lib/fr24/public_html/layers.js has been updated
s6-rc: warning: unable to start service confd-fr24feed: command exited 1
s6-rc: warning: unable to start service confd-piaware: command exited 1
s6-rc: info: service confd-layers successfully started

Restarting piaware from console : 
root@71efd9d24500:/# service piaware --full-restart
Stopping piaware server: piaware.
Starting piaware server: piawarewarning: /etc/piaware.conf: failed to read config file: error reading "file3": illegal operation on a directory

Thom-x, 2023-01-27:
You are mounting some directory into the container. You should not do that.

```
2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[98]: FATAL rename /etc/.piaware.conf636670703 /etc/piaware.conf: file exists

2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[95]: FATAL rename /etc/.fr24feed.ini800013444 /etc/fr24feed.ini: file exists

2023-01-27T10:16:13+01:00 71efd9d24500 /opt/confd/bin/confd[91]: FATAL rename /usr/lib/fr24/public_html/.config.js963024346 /usr/lib/fr24/public_html/config.js: file exists
```

These files are mounted as a directory or a file, the container cannot start because it tries to update the config files but they already exist.
Please review the UNRAID conf so it does not mount unnecessary folder/files.

miodzicho, 2023-01-27:
It seems I have to remove everything in unraid template, and using only extra parameters : --device=$(readlink -f /dev/flightradar) -e "HTML_SITE_LAT=48.xxx"  -e "HTML_SITE_LON=11.xxx" -e "FR24FEED_FR24KEY=key_here" -e "PIAWARE_FEEDER_DASH_ID=id_here" -e "SERVICE_ENABLE_DUMP1090=enable" -e "SERVICE_ENABLE_HTTP=enable" -e "SERVICE_ENABLE_FR24FEED=enable" -e "SERVICE_ENABLE_PIAWARE=enable" --tmpfs /run:exec,size=32M --tmpfs /planefinder/log:exec,size=32M --tmpfs /usr/lib/fr24/public_html/data:size=32M

Thom-x, 2023-01-27:
Perfect 👌👌

closed 2023-01-27 by miodzicho