Thom-x/docker-fr24feed-piaware-dump1090 #110: ADS-B Exchange connection frequency

KingKarlo, 2023-01-30:
Hi, I'm not sure where to ask that question, but I'm sure here are the experts.

Why is the ADS-B Exchange service way more "chatty" than the other services? Is there a way to limit the connection frequency?

Of course, data has to be transferred back to the service, but I see ADS-B like every 5 seconds in my connection and DNS logs, while I see connections to for example Plane Finder(pfclient-upload.planefinder.net) only every 30 seconds.

Is that just my perception, or is my configuration not correct, or is it just the way it is with ADS-B Exchange?

Thank you so much for all the great work!

Thom-x, 2023-02-01:
I guess it could be tweaked to send data with at a different frequency.
But I don't know which config need to be updated on ADS-B Exchange service.
Furthermore, we don't have any way of changing it for this docker image right now, but if we found the proper one we could.

Let's explore ADS-B Exchange config.

KingKarlo, 2023-02-02:
Super cool, thank you very much!

xEtarusx, 2023-04-07:
Can confirm that adsbexchange has the highest rate of dns requests. Planefinder does talk a lot too, but not as much as adsbexchange.
Would appreciate both of them were at the same dns request level of flightradar24.

Here is a screenshot from PiHole with those 3 providers beeing in my "Top Permitted Domains" list:
[image]

t288msd, 2023-06-26:
Some insights from ADSB-exchange devs in this thread.
https://www.reddit.com/r/ADSB/comments/cxz2cd/i_dont_trust_adsb_exchange_i_dont_think_you/

Thom-x, 2023-10-29:
Explanation from Reddit is pretty good. I'll close the issue, I cannot do anything on my side anyway.

closed 2023-10-29 by Thom-x

KingKarlo, 2023-10-29:
Agree! Funny that you close it today, yesterday I decided to disable ADS-B Exchange. I'll only share to other services now.

Planefinder also seems to have a pretty "unique" DNS lookup mechanism, since I have the same behavior as @xEtarusx. But they do only 1/4th of the DNS requests of ADSB-Exchange. ADSB-Exchange did around 36.000 on Friday, more like one every 3 seconds and not "makes DNS request at most twice every 30 seconds" as mentioned on Reddit. So I'm just annoyed by their software 😁 

Thank you for looking into it!