Thom-x/docker-fr24feed-piaware-dump1090 #119: Stopped  working and its in a restart loop

joaoferreira-git, 2023-05-02:
## Specifications

  - Image hash or version: Latest
  - Platform: Unraid
  - Subsystem: Unraid 6.10.3 

## Expected Behavior
Start normally

## Actual Behavior
Restart loop

## Steps to Reproduce the Problem

  1.start container
  
 No obvious errors in the log besides `./run: line 8: ulimit: open files: cannot modify limit: Operation not permitted`

 After a quick google search this seems like a problem with too many files open in a non-root user and after enabling privileged in the docker template it starts without any problem.

 Currently running with root until a fix is available.
  
```[fr24feed] 2023-05-02 16:44:44 | ______  _  _         _      _                    _              _____    ___ 
[fr24feed] 2023-05-02 16:44:44 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |
[fr24feed] 2023-05-02 16:44:44 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
[fr24feed] 2023-05-02 16:44:44 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
[fr24feed] 2023-05-02 16:44:44 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
[fr24feed] 2023-05-02 16:44:44 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
[fr24feed] 2023-05-02 16:44:44 |                __/ |                                                         
[fr24feed] 2023-05-02 16:44:44 |               |___/                                                          
[fr24feed] 2023-05-02 16:44:44 | [main][i]FR24 Feeder/Decoder
[fr24feed] 2023-05-02 16:44:44 | [main][i]Version: 1.0.34-0/generic
[fr24feed] 2023-05-02 16:44:44 | [main][i]Built on Jan 10 2023 10:09:01 (decoder-v1.0.34-0-7c5301b.git/Linux/static_amd64)
[fr24feed] 2023-05-02 16:44:44 | [main][i]Running on: debian="11"
[fr24feed] 2023-05-02 16:44:44 | [main][i]Local IP(s): 172.26.0.13
[fr24feed] 2023-05-02 16:44:44 | [main][i]Copyright 2012-2023 Flightradar24 AB
[fr24feed] 2023-05-02 16:44:44 | [main][i]https://www.flightradar24.com
[fr24feed] 2023-05-02 16:44:44 | [main][i]DNS mode: PING
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service dump1090 successfully stopped
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service dump1090: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service dump1090 successfully started
s6-rc: info: service netcat: starting
s6-rc: info: service netcat successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service init-upintheair: starting
s6-rc: info: service confd-piaware: starting
s6-rc: info: service confd-layers: starting
s6-rc: info: service confd-html: starting
s6-rc: info: service confd-fr24feed: starting
s6-rc: info: service conf-planefinder: starting
s6-rc: info: service conf-adsbexchange: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service conf-planefinder successfully started
s6-rc: info: service planefinder: starting
s6-rc: info: service init-upintheair successfully started
s6-rc: info: service confd-layers successfully started
s6-rc: info: service conf-adsbexchange successfully started
s6-rc: info: service adsbexchange-feed: starting
s6-rc: info: service confd-piaware successfully started
s6-rc: info: service piaware: starting
s6-rc: info: service confd-html successfully started
s6-rc: info: service http: starting
s6-rc: info: service confd-fr24feed successfully started
s6-rc: info: service fr24feed: starting
[... 715 lines cut ...]
s6-rc: info: service http successfully stopped
s6-rc: info: service init-upintheair: stopping
s6-rc: info: service confd-layers: stopping
s6-rc: info: service confd-html: stopping
s6-rc: info: service planefinder-log successfully stopped
s6-rc: info: service planefinder: stopping
s6-rc: info: service netcat successfully stopped
s6-rc: info: service fr24feed successfully stopped
s6-rc: info: service confd-fr24feed: stopping
s6-rc: info: service piaware successfully stopped
s6-rc: info: service confd-piaware: stopping
s6-rc: info: service init-upintheair successfully stopped
s6-rc: info: service confd-layers successfully stopped
s6-rc: info: service confd-html successfully stopped
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service adsbexchange-mlat successfully stopped
s6-rc: info: service planefinder successfully stopped
s6-rc: info: service conf-planefinder: stopping
s6-rc: info: service adsbexchange-stats successfully stopped
s6-rc: info: service adsbexchange-feed: stopping
s6-rc: info: service confd-fr24feed successfully stopped
s6-rc: info: service confd-piaware successfully stopped
s6-rc: info: service conf-planefinder successfully stopped
s6-rc: info: service fix-attrs successfully stopped```

Thom-x, 2023-05-02:
Hi, you can try the previous version. We introduced the change of ulimit to fix a bug, but apparently it create a new one for you.

RestNPizza, 2023-05-03:
I am also on unraid (on v6. 11.5 though) and had the same problem with v1. 21 of this docker. I reverted to v1.20 and that fixed the issue for me.

Darkknight33, 2023-05-05:
I'd also like to echo that 1.21 broke the docker on Unraid and reverting to 1.20 got it running again. I don't have the log copied, but there was a connection issue as if some dependent internal service was not running as expected and caused the docker to exit.

Thom-x, 2023-05-08:
You can try with the image thomx/fr24feed-piaware:PR-122
I disabled ulimit by default, people that need to have the fix need to pass the parameter now.

joaoferreira-git, 2023-05-08:
> You can try with the image thomx/fr24feed-piaware:PR-122 I disabled ulimit by default, people that need to have the fix need to pass the parameter now.

Just confirmed it and its working like normal again without the need for root.
Thanks for the fast update.

RestNPizza, 2023-05-08:
Can also confirm that things are working again for me on PR-122

closed 2023-05-12 by Thom-x