Thom-x/docker-fr24feed-piaware-dump1090 #123: Feature request - add adsb.fi to the list of feeders

crispy78, 2023-05-14:
## Specifications

  - Image hash or version: 1.21
  - Platform: Home Assistant (https://github.com/MaxWinterstein/homeassistant-addons/tree/main/adsb-multi-portal-feeder)
  - Subsystem:

## Feature request

Since there is a lot of debate going online about the (controversial) sale of ADSBExchange.com. 
Would it be possible to add adsb.fi to the list of feeders? And would it also be possible to have both of them added to the default trackers? ADSBExchange.com and adsb.fi allow to view military aircraft by pressing the U.

yousaf465, 2023-06-08:
https://github.com/adsbfi

yousaf465, 2023-06-16:
@Shariner, if you help with this too?

Thom-x, 2023-06-19:
Should be working with image `PR-132`
Minimal config to add:
```
      - "SERVICE_ENABLE_ADSBFI=true"
      - "ADSBFI_UUID=XXX" # generated from `cat /proc/sys/kernel/random/uuid`
      - "ADSBFI_STATION_NAME=XXX"
```

yousaf465, 2023-06-19:
Waiting for the version to update @MaxWinterstein repo also needs to be updated.

closed 2023-06-20 by Thom-x

MaxWinterstein, 2023-06-21:
> [1.26.1] SERVICE_ENABLE_ADSBFI is missing in README : Configure : Common list of services

#136