Thom-x/docker-fr24feed-piaware-dump1090 #131: Radarbox feeder

yousaf465, 2023-06-16:
carrying on from

Thom-x, 2023-06-21:
I'm currently testing it, see https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/pull/135, image tag `PR-135`

closed 2023-06-23 by Thom-x

yousaf465, 2023-06-24:
```➜  ~ docker run -it --rm \
        -e "SERVICE_ENABLE_RADARBOX=true" \
        -e "SERVICE_ENABLE_DUMP1090=false" \
        -e "SERVICE_ENABLE_HTTP=false" \
        -e "SERVICE_ENABLE_PIAWARE=false" \
        -e "SERVICE_ENABLE_FR24FEED=false" \
        -e "HTML_SITE_LAT=45" \
        -e "HTML_SITE_LON=9" \
        -e "HTML_SITE_ALT=0" \
        thomx/fr24feed-piaware /bin/bash
s6-rc: info: service dump1090: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service init-upintheair: starting
s6-rc: info: service confd-piaware: starting
s6-rc: info: service confd-layers: starting
s6-rc: info: service confd-html: starting
s6-rc: info: service confd-fr24feed: starting
s6-rc: info: service conf-planefinder: starting
s6-rc: info: service conf-opensky: starting
s6-rc: info: service conf-adsbexchange: starting
s6-rc: info: service dump1090 successfully started
s6-rc: info: service netcat: starting
s6-rc: info: service confd-piaware successfully started
s6-rc: info: service piaware: starting
s6-rc: info: service confd-layers successfully started
s6-rc: info: service confd-fr24feed successfully started
s6-rc: info: service fr24feed: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service init-upintheair successfully started
s6-rc: info: service confd-html successfully started
s6-rc: info: service http: starting
s6-rc: info: service conf-planefinder successfully started
s6-rc: info: service planefinder: starting
s6-rc: info: service conf-opensky successfully started
s6-rc: info: service opensky: starting
s6-rc: info: service conf-adsbexchange successfully started
s6-rc: info: service adsbexchange-feed: starting
s6-rc: info: service netcat successfully started
s6-rc: info: service piaware successfully started
s6-rc: info: service fr24feed successfully started
s6-rc: info: service http successfully started
s6-rc: info: service planefinder successfully started
s6-rc: info: service planefinder-log: starting
s6-rc: info: service adsbexchange-feed successfully started
s6-rc: info: service adsbexchange-stats: starting
s6-rc: info: service adsbexchange-mlat: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service opensky successfully started
s6-rc: info: service planefinder-log successfully started
s6-rc: info: service adsbexchange-stats successfully started
s6-rc: info: service adsbexchange-mlat successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
root@f55c0798a90d:/# docker run -it --rm \

```
got this when I first ran the radarbox on HA. 

@MaxWinterstein

MaxWinterstein, 2023-06-24:
@yousaf465 try it with the correct image tag? `thomx/fr24feed-piaware:1.27.0`?

yousaf465, 2023-06-24:
> docker run -it --rm \
>         -e "SERVICE_ENABLE_RADARBOX=true" \
>         -e "SERVICE_ENABLE_DUMP1090=false" \
>         -e "SERVICE_ENABLE_HTTP=false" \
>         -e "SERVICE_ENABLE_PIAWARE=false" \
>         -e "SERVICE_ENABLE_FR24FEED=false" \
>         -e "HTML_SITE_LAT=45" \
>         -e "HTML_SITE_LON=9" \
>         -e "HTML_SITE_ALT=0" \
>         thomx/fr24feed-piaware:1.27.0 /bin/bash

you mean like this? 
@MaxWinterstein

MaxWinterstein, 2023-06-24:
Yes. This way your Lokal docker would not reuse an old image as it looks like.

yousaf465, 2023-06-24:
> 

let me try

yousaf465, 2023-06-24:
even waiting for half hour didn't resolved it 
```

[radarbox-feeder] [2023-06-24 23:04:17]  Data received: 629 B
[radarbox-feeder] [2023-06-24 23:04:44]  Connection established.
[radarbox-feeder] [2023-06-24 23:04:44]  Empty sharing key. We will try to create a new one for you!
[radarbox-feeder] [2023-06-24 23:04:54]  Timeout waiting for new key. Will try again in 30 seconds.
[radarbox-feeder] [2023-06-24 23:05:18]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-24 23:05:18]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 27
[radarbox-feeder] [2023-06-24 23:05:18]  Data sent: 675 B
[radarbox-feeder] [2023-06-24 23:05:18]  Data received: 629 B

```

MaxWinterstein, 2023-06-26:
Those key generation is somehow highly unstable to me. I was able to generate keys using some other image on my

**M1 MacBook**
```bash
╰─ timeout 300s docker run \
    --rm \
    -it \
    -e BEASTHOST=ADSBerry \
    -e LAT=YOURLATITUDE \
    -e LONG=YOURLONGITUDE \
    -e ALT=YOURALTITUDE \
    ghcr.io/sdr-enthusiasts/docker-radarbox:latest
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing...
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-show-rbfeeder-changelog.sh: executing...

rbfeeder (1:1.0.8+deb11) bullseye; urgency=medium

  * Fix altitude wrong value.
  * Update dump1090 to the latest version.
  * Filter calculated weather values.
  * New field: TAS.
  * New field: Track.
  * New field: True Heading.
  * New field: ADS-B Version.
  * New field: ADS-R Version.
  * New field: TIS-B Version.
  * New field: MRAR Wind Speed.
  * New field: MRAR Wind Dir.
  * New field: MRAR Pressure.
  * New field: MRAR Temperature.
  * New field: MRAR Turbulence.
  * New field: MRAR Humidity.
  * New field: Roll.
  * New command line option: --debug-level (override configuration file).
  * New command line option: --debug-filter <text> to filter debug messages.

 --  Jonis Maurin Ceara <jceara@airnavsystems.com>  Fri, 08 Jul 2022 16:52:56 +0300

[cont-init.d] 01-show-rbfeeder-changelog.sh: exited 0.
[cont-init.d] 02-rbfeeder.sh: executing...
WARNING: TZ environment variable not set

WARNING: SHARING_KEY environment variable was not set!
Please make sure you note down the key generated.
Pass the key as environment var SHARING_KEY on next launch!

[cont-init.d] 02-rbfeeder.sh: exited 0.
[cont-init.d] 03-show-architecture.sh: executing...

Hardware information:
Machine:   aarch64
Processor: unknown
Platform:  unknown

[cont-init.d] 03-show-architecture.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[rbfeeder] [2023-06-26 14:41:04]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2023-06-26 14:41:04]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2023-06-26 14:41:04]  Network-mode enabled.
[rbfeeder] [2023-06-26 14:41:04]  		Remote host to fetch data: 192.168.1.125
[rbfeeder] [2023-06-26 14:41:04]  		Remote port: 30005
[rbfeeder] [2023-06-26 14:41:04]  		Remote protocol: BEAST
[rbfeeder] [2023-06-26 14:41:04]  Using GNSS (when available)
[rbfeeder] [2023-06-26 14:41:04]  Start date/time: 2023-06-26 14:41:04
[rbfeeder] [2023-06-26 14:41:04]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] [2023-06-26 14:41:05]  Connection established.
[rbfeeder] [2023-06-26 14:41:05]  Empty sharing key. We will try to create a new one for you!
[rbfeeder] [2023-06-26 14:41:05]  CPU Serial empty. Use MAC address instead.
[rbfeeder] [2023-06-26 14:41:05]  Your new key is ddec4b179c2abf1b9XXXXXX927228b54a. Please save this key for future use. You will have to know this key to link this receiver to your account in RadarBox24.com. This key is also saved in configuration file (/etc/rbfeeder.ini)
^C[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
^C[s6-finish] sending all processes the TERM signal.
^C[s6-finish] sending all processes the KILL signal and exiting.
```

But not on my **orangepi**. Sadly no Raspberry PI here at the moment.

Shariner, 2023-06-26:
First: amazing fast work.

I am also getting errors with my current sharing key, it seems not being valid and after a few tries, the server accepted it and the station was successfully authenticated to my account. I will monitor this if it will fail or have problems with the integration/authentication. Thanks. 

```shell
[radarbox-feeder] [2023-06-26 16:36:49]  Connection established.
[radarbox-feeder] [2023-06-26 16:36:49]  CPU Serial empty. Use MAC address instead.
[radarbox-feeder] [2023-06-26 16:36:49]  Error authenticating Sharing-Key: Invalid sharing-key
[radarbox-feeder] [2023-06-26 16:36:59]  Could not start connection. Timeout.
[fr24feed] 2023-06-26 16:37:04 | [feed][n]ping 5
[fr24feed] 2023-06-26 16:37:05 | [feed][n]syncing stream result: 1
[radarbox-feeder] [2023-06-26 16:37:09]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-26 16:37:09]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 5
[radarbox-feeder] [2023-06-26 16:37:09]  Data sent: 315 B
[radarbox-feeder] [2023-06-26 16:37:09]  Data received: 135 B
[fr24feed] 2023-06-26 16:37:34 | [feed][n]ping 6
[fr24feed] 2023-06-26 16:37:35 | [feed][n]syncing stream result: 1
[piaware] 3 msgs recv'd from dump1090 (3 in last 5m); 3 msgs sent to FlightAware
[radarbox-feeder] [2023-06-26 16:38:00]  Connection established.
[radarbox-feeder] [2023-06-26 16:38:00]  CPU Serial empty. Use MAC address instead.
[radarbox-feeder] [2023-06-26 16:38:01]  Client type: Raspberry Pi
[radarbox-feeder] [2023-06-26 16:38:01]  Connection with RadarBox24 server OK! Key accepted by server.
[radarbox-feeder] [2023-06-26 16:38:01]  This is your station serial number: EXTRPIXXXXXX
[fr24feed] 2023-06-26 16:38:04 | [feed][n]ping 7
[fr24feed] 2023-06-26 16:38:05 | [feed][n]syncing stream result: 1
[radarbox-feeder] [2023-06-26 16:38:10]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-26 16:38:10]  Packets sent in the last 60 seconds: 2, Total packets sent since startup: 7
[radarbox-feeder] [2023-06-26 16:38:10]  Data sent: 494 B
[radarbox-feeder] [2023-06-26 16:38:10]  Data received: 166 B
[fr24feed] 2023-06-26 16:38:34 | [feed][n]ping 8
[fr24feed] 2023-06-26 16:38:35 | [feed][n]syncing stream result: 1
[fr24feed] 2023-06-26 16:39:04 | [feed][n]ping 9
[fr24feed] 2023-06-26 16:39:05 | [feed][n]syncing stream result: 1
[radarbox-feeder] [2023-06-26 16:39:11]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-26 16:39:11]  Packets sent in the last 60 seconds: 3, Total packets sent since startup: 10
[radarbox-feeder] [2023-06-26 16:39:11]  Data sent: 519 B
[radarbox-feeder] [2023-06-26 16:39:11]  Data received: 166 B
[fr24feed] 2023-06-26 16:39:34 | [feed][n]ping 10
[fr24feed] 2023-06-26 16:39:35 | [feed][n]syncing stream result: 1
[fr24feed] 2023-06-26 16:40:04 | [feed][n]ping 11
[fr24feed] 2023-06-26 16:40:05 | [feed][n]syncing stream result: 1
[radarbox-feeder] [2023-06-26 16:40:12]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-26 16:40:12]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 11
[radarbox-feeder] [2023-06-26 16:40:12]  Data sent: 526 B
[radarbox-feeder] [2023-06-26 16:40:12]  Data received: 166 B
[fr24feed] 2023-06-26 16:40:34 | [feed][n]ping 12
[fr24feed] 2023-06-26 16:40:35 | [feed][n]syncing stream result: 1
``` 

I am using the addon on a usual Lenovo M900 computer with intel processor, so not a raspberry.

Thom-x, 2023-06-26:
@MaxWinterstein @Shariner I read [here](https://github.com/sdr-enthusiasts/docker-radarbox#connection-errors), that is could be unstable on Radarbox side :

> Connection Errors
> Before raising an issue regarding connection errors, please wait at least 10 minutes. The rbfeeder binary is configured to attempt to connect to a collection of servers in a round-robin method. It appears normal for some servers to reject the connection, so it may take several minutes to find an available server and connect.
> 
> You may also receive a spurious error Error authenticating Sharing-Key: Invalid sharing-key. Provided you have entered your sharing key correctly, just ignore this for several minutes.
> 
> Here is some example output with RBFeeder Version 1.0.8 (build 20220708190411) showing the aformentioned behaviour:

Maybe try what is said.

yousaf465, 2023-06-26:
> @MaxWinterstein @Shariner I read [here](https://github.com/sdr-enthusiasts/docker-radarbox#connection-errors), that is could be unstable on Radarbox side :
>
>
> Maybe try what is said.

I let it run for more than 3 hours last time from 3am to almost 6AM. same result.

Thom-x, 2023-06-26:
Ok, doesn't seems to be this exact issue. I really don't know what to do.

yousaf465, 2023-07-07:
finally radarbox is working for me

```
[radarbox-feeder] [2023-07-08 02:14:01]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-07-08 02:14:01]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 4
[radarbox-feeder] [2023-07-08 02:14:01]  Data sent: 193 B
[radarbox-feeder] [2023-07-08 02:14:01]  Data received: 73 B
```

yousaf465, 2023-07-08:
```[radarbox-feeder] Traceback (most recent call last):
[radarbox-feeder]   File "/usr/local/share/radarbox-mlat-client/venv/bin/mlat-client", line 22, in <module>
[radarbox-feeder]     import mlat.client.version
[radarbox-feeder] ModuleNotFoundError: No module named 'mlat'

``` anyone else facing this issue?

MaxWinterstein, 2023-07-09:
> ```
> [radarbox-feeder]   File "/usr/local/share/radarbox-mlat-client/venv/bin/mlat-client", line 22, in <module>
> [radarbox-feeder]     import mlat.client.version
> [radarbox-feeder] ModuleNotFoundError: No module named 'mlat'
> ```
> 
> anyone else facing this issue?

Can't really test right now, but would like to point to this: https://community.home-assistant.io/t/home-assistant-add-on-ads-b-multi-portal-feeder-flightradar24-and-flightaware/345563/59

MaxWinterstein, 2023-07-09:
@Thom-x I am not sure, but might this lead towards the wrong Python? 

https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/912c641949fc2fe7571cf0be47443ca8cb15b7b0/root/etc/s6-overlay/s6-rc.d/conf-radarbox/script#L53

```python
root@4804ada279cf:/# /usr/bin/python3 /usr/local/share/radarbox-mlat-client/venv/bin/mlat-client
Traceback (most recent call last):
  File "/usr/local/share/radarbox-mlat-client/venv/bin/mlat-client", line 22, in <module>
    import mlat.client.version
ModuleNotFoundError: No module named 'mlat'
```

But running it like this:
```python
root@4804ada279cf:/# /usr/local/share/radarbox-mlat-client/venv/bin/python3 /usr/local/share/radarbox-mlat-client/venv/bin/mlat-client
usage: mlat-client [-h] [--input-type {auto,dump1090,beast,radarcape_12mhz,radarcape_gps,radarcape,sbs,avrmlat}] --input-connect INPUT_CONNECT [--results RESULTS]
                   [--no-anon-results] [--no-modeac-results] --lat LAT --lon LON --alt ALT [--privacy] --user USER [--server SERVER] [--no-udp]
mlat-client: error: the following arguments are required: --input-connect, --lat, --lon, --alt, --user
```

seems to do the job?

Thom-x, 2023-07-09:
Yes you are right, I did it for testing https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/912c641949fc2fe7571cf0be47443ca8cb15b7b0/Dockerfile#L404
But completely forgot to change here...
Let's fix that.

Thom-x, 2023-07-09:
https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/pull/138