Thom-x/docker-fr24feed-piaware-dump1090 #133: Segmentation fault on Raspberry Pi Zero W

daveyijzermans, 2023-06-16:
## Specifications

  - Image hash or version: 1.25.0
  - Platform: Raspberry Pi Zero W (armv6l)
  - Subsystem:

## Expected Behavior

Planefinder and opensky feeder start successfully

## Actual Behavior

Segmentation faults appear in console when either opensky or planefinder feeder is enabled

```
./run: line 21:   146 Segmentation fault      (core dumped) /planefinder/pfclient --connection_type=1 --address="127.0.0.1" --port="30005" --data_format=1 --sharecode="${PLANEFINDER_SHARECODE}" --lat="${HTML_SITE_LAT}" --lon="${HTML_SITE_LON}" --pid_file=/run/pfclient.pid --config_path=/planefinder/config/pfclient-config.json --log_path=/planefinder/log 2>&1
       147 Done                    | mawk -W interactive '{printf "%c[36m[planefinder]%c[0m %s\n", 27, 27, $0}'
```
```
./run: line 9:   145 Segmentation fault      (core dumped) /usr/bin/openskyd-dump1090 2>&1
       146 Done                    | mawk -W interactive '{printf "%c[34m[opensky-feeder]%c[0m %s\n", 27, 27, $0}'
```

I am not sure if this is expected on this platform (Raspberry Pi Zero W). It's probably underpowered and I'm not sure if these feeders are even supposed to run on RPi Zero. I'm running all other feeders on it no problem though.

## Steps to Reproduce the Problem

  1. Enable either the opensky or planefinder feeder

Thom-x, 2023-06-17:
It might probably what you say. I have a Pi Zero W, and I'm only running dump1090 on it, pf was failing so I started the other service on another computer (amd64).

You can try to disable the service and only run `/planefinder/pfclient --version` manually. If it fails we'll amost be sure we cannot do anything about it.

daveyijzermans, 2023-06-17:
You're right, it also displays the Segmentation fault, so I guess these feeders do not run on a Zero.

I've setup the Pi to only get the dump1090 and to feed that to another server that does the feeding using your example config [here](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/96#issuecomment-1174937202).

Thanks for making such a versatile and easy all in one image!

closed 2023-06-17 by daveyijzermans