Thom-x/docker-fr24feed-piaware-dump1090 #141: STATUS: Exited (0)

ErikProductions, 2023-08-25:
I'm trying to run it in docker on a raspberry pi 4. After executing the run command the container successfully starts, but then the STATUS changes to Exited (0). 

Any help would be appreciated.

ErikProductions, 2023-08-25:
I'm using this command:

docker run -d -p 8080:8080 -p 8754:8754 --name dump\
 --device=/dev/bus/usb:/dev/bus/usb \
 -v "/etc/localtime:/etc/localtime:ro" \
 -e "HTML_SITE_LAT=MY_SITE_LAT" \
 -e "HTML_SITE_LON=MY_SITE_LON" \
 -e "HTML_SITE_NAME=MY_SITE_NAME" \
 -e "MLAT_EXACT_LAT=MY_EXACT_SITE_LAT" \
 -e "MLAT_EXACT_LON=MY_EXACT_SITE_LON" \
 -e "MLAT_ALTITUDE_MSL_METERS=MY_SITE_ALT_MSL_METERS" \
 --tmpfs /run:exec,size=32M \
 thomx/fr24feed-piaware

(I have very little experience with docker)

yayitazale, 2023-08-28:
Same here, in the logs i don't see anything wrong...

```
2023-08-28T09:00:40.825410628Z [fr24feed] 2023-08-28 09:00:40 | ______  _  _         _      _                    _              _____    ___ 
2023-08-28T09:00:40.825565148Z [fr24feed] 2023-08-28 09:00:40 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |
2023-08-28T09:00:40.825603061Z [fr24feed] 2023-08-28 09:00:40 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
2023-08-28T09:00:40.825634361Z [fr24feed] 2023-08-28 09:00:40 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
2023-08-28T09:00:40.825664776Z [fr24feed] 2023-08-28 09:00:40 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
2023-08-28T09:00:40.825699825Z [fr24feed] 2023-08-28 09:00:40 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
2023-08-28T09:00:40.825731281Z [fr24feed] 2023-08-28 09:00:40 |                __/ |                                                         
2023-08-28T09:00:40.825761331Z [fr24feed] 2023-08-28 09:00:40 |               |___/                                                          
2023-08-28T09:00:40.829575053Z [fr24feed] 2023-08-28 09:00:40 | [main][i]FR24 Feeder/Decoder
2023-08-28T09:00:40.829709574Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Version: 1.0.34-0/generic
2023-08-28T09:00:40.829744259Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Built on Jan 10 2023 10:02:35 (decoder-v1.0.34-0-7c5301b.git/Linux/static_armel)
2023-08-28T09:00:40.837748705Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Running on: debian="11"
2023-08-28T09:00:40.838841956Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Local IP(s): 172.18.0.2
2023-08-28T09:00:40.838977310Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Copyright 2012-2023 Flightradar24 AB
2023-08-28T09:00:40.839012047Z [fr24feed] 2023-08-28 09:00:40 | [main][i]https://www.flightradar24.com
2023-08-28T09:00:40.839042306Z [fr24feed] 2023-08-28 09:00:40 | [main][i]DNS mode: PING
2023-08-28T09:00:40.873321119Z [fr24feed] 2023-08-28 09:00:40 | [main][i]Automatic updates are DISABLED
2023-08-28T09:00:40.874878918Z [fr24feed] 2023-08-28 09:00:40 | 23-08-28 09:00:40.867 [I][http-server.cpp:270] [httpd]Server started, listening on :::8754
2023-08-28T09:00:40.892959322Z s6-rc: info: service legacy-services successfully started
2023-08-28T09:00:40.894965941Z s6-rc: info: service legacy-services: stopping
2023-08-28T09:00:40.910873280Z s6-rc: info: service legacy-services successfully stopped
2023-08-28T09:00:40.922293666Z s6-rc: info: service legacy-cont-init: stopping
2023-08-28T09:00:40.930468725Z s6-rc: info: service radarbox: stopping
2023-08-28T09:00:40.938324641Z s6-rc: info: service planefinder-log: stopping
2023-08-28T09:00:40.944816546Z s6-rc: info: service piaware: stopping
2023-08-28T09:00:40.950906554Z s6-rc: info: service opensky: stopping
2023-08-28T09:00:40.960362922Z s6-rc: info: service netcat: stopping
2023-08-28T09:00:40.967215998Z s6-rc: info: service http: stopping
2023-08-28T09:00:40.969935844Z s6-rc: info: service fr24feed: stopping
2023-08-28T09:00:40.979660369Z s6-rc: info: service adsbfi-mlat: stopping
2023-08-28T09:00:40.986164877Z s6-rc: info: service adsbexchange-stats: stopping
2023-08-28T09:00:40.993756907Z s6-rc: info: service adsbexchange-mlat: stopping
2023-08-28T09:00:40.993897001Z s6-rc: info: service planefinder-log successfully stopped
2023-08-28T09:00:40.999409084Z s6-rc: info: service planefinder: stopping
2023-08-28T09:00:41.003610277Z s6-rc: info: service radarbox successfully stopped
2023-08-28T09:00:41.011148730Z s6-rc: info: service conf-radarbox: stopping
2023-08-28T09:00:41.023514226Z s6-rc: info: service piaware successfully stopped
2023-08-28T09:00:41.034076835Z s6-rc: info: service confd-piaware: stopping
2023-08-28T09:00:41.043490882Z s6-rc: info: service netcat successfully stopped
2023-08-28T09:00:41.049617775Z s6-rc: info: service adsbfi-mlat successfully stopped
2023-08-28T09:00:41.053927145Z s6-rc: info: service adsbfi-feed: stopping
2023-08-28T09:00:41.055430107Z s6-rc: info: service conf-radarbox successfully stopped
2023-08-28T09:00:41.063498680Z s6-rc: info: service confd-piaware successfully stopped
2023-08-28T09:00:41.064731611Z s6-rc: info: service http successfully stopped
2023-08-28T09:00:41.077475985Z s6-rc: info: service init-upintheair: stopping
2023-08-28T09:00:41.083459087Z s6-rc: info: service confd-layers: stopping
2023-08-28T09:00:41.087321719Z s6-rc: info: service confd-html: stopping
2023-08-28T09:00:41.087592741Z s6-rc: info: service legacy-cont-init successfully stopped
2023-08-28T09:00:41.090270044Z s6-rc: info: service fix-attrs: stopping
2023-08-28T09:00:41.090486903Z s6-rc: info: service opensky successfully stopped
2023-08-28T09:00:41.094158039Z s6-rc: info: service conf-opensky: stopping
2023-08-28T09:00:41.094299383Z s6-rc: info: service fr24feed successfully stopped
2023-08-28T09:00:41.101788936Z s6-rc: info: service confd-fr24feed: stopping
2023-08-28T09:00:41.102419255Z s6-rc: info: service adsbexchange-stats successfully stopped
2023-08-28T09:00:41.102538204Z s6-rc: info: service adsbexchange-mlat successfully stopped
2023-08-28T09:00:41.104638310Z s6-rc: info: service adsbexchange-feed: stopping
2023-08-28T09:00:41.104860273Z s6-rc: info: service confd-html successfully stopped
2023-08-28T09:00:41.105580012Z s6-rc: info: service fix-attrs successfully stopped
[... cut ...]
2023-08-28T09:00:41.150312137Z s6-rc: info: service conf-adsbfi successfully stopped
2023-08-28T09:00:41.150344687Z s6-rc: info: service conf-planefinder successfully stopped
2023-08-28T09:00:41.155530451Z s6-rc: info: service conf-adsbexchange successfully stopped
2023-08-28T09:00:41.156185455Z s6-rc: info: service s6rc-oneshot-runner: stopping
2023-08-28T09:00:41.157176997Z s6-rc: info: service dump1090 successfully stopped
2023-08-28T09:00:41.164607023Z s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

ErikProductions, 2023-08-28:
Eskerrik asko! :) I just followed the tutorial on your blog.

yayitazale, 2023-08-28:
> Same here, in the logs i don't see anything wrong...
>
> ```
> 2023-08-28T09:00:40.825410628Z [fr24feed] 2023-08-28 09:00:40 | ______  _  _         _      _                    _              _____ [...]

With a few RPI restarts I see this, so maybe my RTL dongle is dyeing:

```
2023-08-28T09:36:18.063062332Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:18 +0000] "GET /data-978/receiver.json?_=1693215377826 HTTP/1.1" 404 0 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Win
dows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:18.063458166Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:18 +0000] "GET /data/receiver.json?_=1693215377827 HTTP/1.1" 200 94 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:18.135327541Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:18 +0000] "GET /data/history_0.json?_=1693215377828 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Windo
ws NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:18.225246134Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:18 +0000] "GET /data/aircraft.json?_=1693215377829 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:19.213099311Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:19 +0000] "GET /data/aircraft.json?_=1693215377830 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:20.225130196Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:20 +0000] "GET /data/aircraft.json?_=1693215377831 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:20.788974884Z [fr24feed] 2023-08-28 09:36:20 | [time][i]Synchronizing time via NTP
2023-08-28T09:36:20.957342019Z [fr24feed] 2023-08-28 09:36:20 | [time][i]Time synchronized correctly, offset -0.002 seconds
2023-08-28T09:36:20.963814311Z [fr24feed] 2023-08-28 09:36:20 | [feed][i]Downloading configuration
2023-08-28T09:36:20.963925456Z [fr24feed] 2023-08-28 09:36:20 | [main][i]Feed Network client started
2023-08-28T09:36:20.975533998Z [fr24feed] 2023-08-28 09:36:20 | [feed][d]fetching configuration
2023-08-28T09:36:21.234262733Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:21 +0000] "GET /data/aircraft.json?_=1693215377832 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:21.258796919Z [fr24feed] 2023-08-28 09:36:21 | [feed][i]configuring decoder
2023-08-28T09:36:21.271600912Z [fr24feed] 2023-08-28 09:36:21 | [feed][c]Max range AIR: 350.0nm
2023-08-28T09:36:21.275866323Z [fr24feed] 2023-08-28 09:36:21 | [feed][c]Max range GND: 80.0nm
2023-08-28T09:36:21.281745270Z [fr24feed] 2023-08-28 09:36:21 | [feed][i]configuration changed
2023-08-28T09:36:21.286151942Z [fr24feed] 2023-08-28 09:36:21 | [feed][i]defined 3 servers
2023-08-28T09:36:21.288688331Z [fr24feed] 2023-08-28 09:36:21 | [feed][c]Timestamps: optional
2023-08-28T09:36:21.291048314Z [fr24feed] 2023-08-28 09:36:21 | 23-08-28 09:36:21.272 [I][crxstats.cpp:588] [stats]Stats thread started
2023-08-28T09:36:21.296644428Z [fr24feed] 2023-08-28 09:36:21 | 23-08-28 09:36:21.276 [I][receiver_ac_sender.cpp:137] Stopping ReceiverACSender threads for feed 
2023-08-28T09:36:21.302820579Z [fr24feed] 2023-08-28 09:36:21 | 23-08-28 09:36:21.276 [D][receiver_ac_sender.cpp:141] Stop called on non-running network thread for feed 
2023-08-28T09:36:21.308957262Z [fr24feed] 2023-08-28 09:36:21 | 23-08-28 09:36:21.276 [I][receiver_ac_sender.cpp:96] Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.21
8.24.24:8099, feed: LESO15, send_interval: 5s/1s, max age: 15s, send metadata: true, mode: 1, filtering: true
2023-08-28T09:36:21.317941181Z [fr24feed] 2023-08-28 09:36:21 | [feed][n]LESO15@185.218.24.22:8099/UDP
2023-08-28T09:36:21.325937786Z [fr24feed] 2023-08-28 09:36:21 | 23-08-28 09:36:21.280 [I][receiver_ac_sender.cpp:36] Network thread connecting to 185.218.24.22:8099 for feed LESO15
2023-08-28T09:36:21.334068675Z [fr24feed] 2023-08-28 09:36:21 | [feed][n]connecting
2023-08-28T09:36:21.394688960Z [fr24feed] 2023-08-28 09:36:21 | [feed][n]connected via UDP (fd 25)
2023-08-28T09:36:21.400839702Z [fr24feed] 2023-08-28 09:36:21 | [feed][n]working
2023-08-28T09:36:22.238531873Z [http] 192.168.1.20 - - [28/Aug/2023:09:36:22 +0000] "GET /data/aircraft.json?_=1693215377833 HTTP/1.1" 200 65 "https://fr.yayitazale.cc/" "Mozilla/5.0 (Window
s NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0"
2023-08-28T09:36:22.807355674Z [dump1090] Mon Aug 28 09:36:22 2023 GMT  No samples received from the SDR for a long time. Maybe the hardware is wedged? Giving up.
2023-08-28T09:36:22.808650108Z [dump1090] Mon Aug 28 09:36:22 2023 GMT  Waiting for receive thread termination
2023-08-28T09:36:22.946464744Z [dump1090] rtlsdr_demod_write_reg failed with -4
2023-08-28T09:36:22.946980903Z [dump1090] rtlsdr_demod_read_reg failed with -4
2023-08-28T09:36:22.947214329Z [dump1090] rtlsdr_demod_write_reg failed with -4
2023-08-28T09:36:22.947325444Z [dump1090] rtlsdr_demod_read_reg failed with -4
2023-08-28T09:36:22.947357987Z [dump1090] rtlsdr_write_reg failed with -4
2023-08-28T09:36:22.947434841Z [dump1090] Reattaching kernel driver failed!
[... cut ...]
2023-08-28T09:36:23.154746130Z [piaware] faup1090 exited normally
2023-08-28T09:36:23.155572479Z [piaware] will reconnect to dump1090 in 30 seconds
2023-08-28T09:36:23.168517568Z s6-rc: info: service legacy-services successfully stopped
2023-08-28T09:36:23.169009368Z s6-rc: info: service legacy-cont-init: stopping
2023-08-28T09:36:23.171531893Z s6-rc: info: service radarbox: stopping
2023-08-28T09:36:23.175061188Z s6-rc: info: service planefinder-log: stopping
2023-08-28T09:36:23.176011516Z s6-rc: info: service piaware: stopping
```

EagleAdam, 2023-08-29:
I am having the same issue albeit on a different system. I’m running this on an odroid N2+. So far I’m trying to figure out if dump1090 can even start and connect to the dongle. From the host OS I ran 

`rtl_test -t’

and I got a result saying that it connected to the device ok. However it almost seems that the docker container cannot connect… and then just shuts down. Is there a way to run the container in a more verbose / debug mode?

yayitazale, 2023-08-29:
I did a rpi-update and now, after several reboots, I can make it run for 4-5 minutes. Then the docker dies.

t288msd, 2023-08-29:
2023-08-28T09:36:22.946464744Z [dump1090] rtlsdr_demod_write_reg failed with -4
2023-08-28T09:36:22.946980903Z [dump1090] rtlsdr_demod_read_reg failed with -4
2023-08-28T09:36:22.947214329Z [dump1090] rtlsdr_demod_write_reg failed with -4
2023-08-28T09:36:22.947325444Z [dump1090] rtlsdr_demod_read_reg failed with -4
2023-08-28T09:36:22.947357987Z [dump1090] rtlsdr_write_reg failed with -4
2023-08-28T09:36:22.947434841Z [dump1090] Reattaching kernel driver failed!

The application can't secure access to the rtl-sdr tuner.

I didn't have this issue but apparently sometimes the pi kernel loads default drivers that need to be blacklisted. try some of the links here: https://www.google.co.uk/search?q=blacklist+kernel+rtl-sdr+pi&ie=utf-8&oe=utf-8&aq=t

Every now and then I get "rtlsdr_demod_write_reg failed with -7" on two different rtl-sdrs with different apps and a reboot fixes.

t288msd, 2023-09-01:
and be sure to follow the instructions here for configuring the pi for rtl-sdr  (scroll to the linux section) 
https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

yayitazale, 2023-09-01:
Why I have to deal with this If I'm using a docker enviroment?

yayitazale, 2023-09-01:
I have just reinstalled a whole RPI OS, docker and docker-compose from scratch and now it is working again.

closed 2023-09-11 by Thom-x