Thom-x/docker-fr24feed-piaware-dump1090 #143: multi feeder

rembulan, 2023-10-23:
can You tell me is that posible to feed multi feeder like flightaware, ADSBexchange etc ?
what hardare requirement let say for 5 site simultanoesly

Thom-x, 2023-10-23:
I'm feeding all available feeders on this image at the same time.
Even a small Raspberry Pi should be able to do it (not pi zero, there are some arch incompatibilies).

Eurotimmy, 2023-10-25:
Hey @rembulan 
I feed everybody all off the one device.

I started by setting the first 4 (below) to 'true' then start subscribing and enabling each service you wish to feed.
The [README.md](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/master/README.md) I think is very thorough to assist with the configurations of each service. 

**FIRST**
SERVICE_ENABLE_DUMP1090
SERVICE_ENABLE_PIAWARE
SERVICE_ENABLE_FR24FEED
SERVICE_ENABLE_HTTP

**NEXT**
SERVICE_ENABLE_ADSBEXCHANGE
SERVICE_ENABLE_PLANEFINDER
SERVICE_ENABLE_OPENSKY
SERVICE_ENABLE_ADSBFI
SERVICE_ENABLE_RADARBOX
SERVICE_ENABLE_ADSBHUB

closed 2023-10-29 by Thom-x