Thom-x/docker-fr24feed-piaware-dump1090 #148: Connection refused at web interface (8080) on 1.29.0

Tezzlicious, 2023-10-29:
## Specifications

  - Image hash or version: 1.29.0
  - Platform: Raspbian GNU/Linux 11 (bullseye) and Docker version 24.0.5
  - Subsystem: Raspberry Pi 3 Model B connected through LAN

## Expected Behavior

Web interface loads up at http://<ip>:8080/

## Actual Behavior

Connection refused at http://<ip>:8080/

## Steps to Reproduce the Problem

  1. On 1.28.0 everything works
  2. Update container to 1.29.0
  3. Connection refused at http://<ip>:8080/. Seems that the data is sent out to feeds.

Reverting to 1.28.0 and the web interface is up again.

Thom-x, 2023-10-29:
Hello, same for me. I'll have a look

Thom-x, 2023-10-29:
Easy fix, of course. 
hank you for spotting this !

closed 2023-10-29 by Thom-x