Thom-x/docker-fr24feed-piaware-dump1090 #150: ADSBHub sometimes key could not work

73-de-LZ, 2023-10-29:
I had an issue with feeding ADSBhub. First i realise the just removing the container and starting with new env vars doesnt update the image, so please add in the manual :latest as here:
`docker run -d -p 8080:8080 -p 8754:8754 \
	--device=/dev/bus/usb:/dev/bus/usb \
.
.
.
	-e "SERVICE_ENABLE_ADSBHUB=true" \
	-e "ADSBHUB_CKEY=MY_ADSBHUB_SHARING_KEY" \
	--tmpfs /run:exec,size=32M \
	--tmpfs /planefinder/log:exec,size=32M \
	--tmpfs /usr/lib/fr24/public_html/data:size=32M \
	thomx/fr24feed-piaware:latest`
Next problem was that mine generated key contains that part:
1.`.........o88{xA94knz$aRSGnjU+{f3;~3J8$WHM).........`
2`.........o88{xA94knz+{f3;~3J8).........`
and after checking real content of env vars with `docker exec -it <container_name> env` it was as on 2.
So it always good to check if there is no feed and your key contains "**$**", just add new station in ADSBhub and delete that one.

Thom-x, 2023-10-29:
Hi, I'll see what we can do to escape special chars, it's always a burden to plays with that kind of issue.

About docker image tag, sorry I won't explain that in the doc, this is very basic docker stuff and you should rather get familiar on how docker tags are working on their doc.

Thom-x, 2023-10-29:
Did you try to pass the environment variable with simple quote?

`-e 'ADSBHUB_CKEY=MY_ADSBHUB_SHARING_KEY'`

If you don't, bash/zsh etc. will substitute the $xxx with the variable value.

73-de-LZ, 2023-10-29:
In fact i was tried simple quote few days back, because had space in some env, but now i did just copy/paste, so no.

As i don't want to open another issue is it will be good to mention here my observation on message rate of dump1090 WebUI? Saw more than twice increased rate, after enabling Radarbox, no matter with or without MLAT. Ususally i have 500msg/s in both dump1090 and tar1090, wich is fed by dump1090, but now when i start Radarbox feeding, at same conditions i get 1300mes/s and this is only in dump1090. On tar1090 and graph1090 rate is my normal 500mes/s. What can cause this and what i could do to fix it or diagnose it, as looking on the logs dosn't help me alot on that.

Thom-x, 2023-10-29:
I'm sorry I won't be any help on that.
My one cent, it doubles the rate because there is another connection done by Radarbox or something like that.

closed 2023-10-30 by Thom-x