Thom-x/docker-fr24feed-piaware-dump1090 #162: Services in container keeps stopping

fl0wfr, 2024-03-08:
## Specifications

  - Image hash or version: Latest sha256:74f48478bed7b28e7e236bfe50515bac9800d0ea2132f10e41731f286c81e971
  - Platform: Rocky Linux 9 + Docker CE 25.0.3
  - Subsystem: x86_64

## Expected Behavior
Services should start without stopping right after.

## Actual Behavior
Services keeps stopping during container start.

## Steps to Reproduce the Problem

  1. Docker command:
```
docker run  --name=fr24feed  \
		--hostname=fr24feed  \
		--env=MLAT_ALTITUDE_MSL_METERS=xxx  \
		--env=FR24FEED_FR24KEY=xxx  \
		--env=PIAWARE_FEEDER_DASH_ID=xxx  \
		--env=MLAT_EXACT_LON=xxx  \
		--env=HTML_SITE_LON=xxx  \
		--env=MLAT_EXACT_LAT=xxx  \
		--env=HTML_SITE_NAME=Home  \
		--env=HTML_SITE_LAT=xxx  \
		--env=TZ=Etc/GMT+0 \
		--volume=/etc/localtime:/etc/localtime:ro  \
		--tmpfs /run:exec,size=32M \
		--tmpfs /planefinder/log:exec,size=32M \
		--tmpfs /usr/lib/fr24/public_html/data:size=32M \
		-p 8083:8080  \
		-p 8754:8754  \
		--restart=unless-stopped  \
		--device /dev/bus/usb:/dev/bus/usb  \
		--log-driver=syslog  \
		--log-opt tag={{.Name}}  \
		--log-opt syslog-address=udp://127.0.0.1:514  \
		--runtime=runc  \
		--detach=true  \
		docker.io/thomx/fr24feed-piaware:latest
```
  2. Check logs:
```
[fr24feed] 2024-03-08 09:00:23 | [main][i]Local IP(s): xxx
[fr24feed] 2024-03-08 09:00:23 | [main][i]Copyright 2012-2023 Flightradar24 AB
[fr24feed] 2024-03-08 09:00:23 | [main][i]https://www.flightradar24.com
[fr24feed] 2024-03-08 09:00:23 | [main][i]DNS mode: PING
[fr24feed] 2024-03-08 09:00:23 | [main][i]Automatic updates are DISABLED
[fr24feed] 2024-03-08 09:00:23 | 24-03-08 09:00:23.008 [I][http-server.cpp:246] [httpd]Server started, listening on :::8754
[dump1090] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2832UHIDIR, SN 00000001)
[piaware] ****************************************************
[piaware] piaware version 9.0.1 is running, process ID 171
[piaware] your system info is: Linux fr24feed 5.14.0-362.18.1.el9_3.0.1.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Feb 11 13:49:23 UTC 2024 x86_64 GNU/Linux
[fr24feed] 2024-03-08 09:00:23 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
[fr24feed] 2024-03-08 09:00:23 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
[fr24feed] 2024-03-08 09:00:23 | [main][i]Reader thread started
[fr24feed] 2024-03-08 09:00:23 | [master][i]Starting processing thread
[fr24feed] 2024-03-08 09:00:23 | [reader][i]Initializing reader
[fr24feed] 2024-03-08 09:00:23 | [reader][i]Connecting to unknown receiver via (tcp://127.0.0.1:30005)
[fr24feed] 2024-03-08 09:00:23 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://127.0.0.1:30005
[fr24feed] 2024-03-08 09:00:23 | [time][i]Synchronizing time via NTP
[dump1090] Found Rafael Micro R820T tuner
[fr24feed] 2024-03-08 08:59:56 | [time][i]Time synchronized correctly, offset -26.497 seconds
[fr24feed] 2024-03-08 08:59:56 | [main][i]Feed Network client started
[fr24feed] 2024-03-08 08:59:56 | [feed][i]Downloading configuration
[fr24feed] 2024-03-08 08:59:56 | [feed][d]Fetching configuration
[fr24feed] 2024-03-08 08:59:56 | [feed][i]configuring decoder
[fr24feed] 2024-03-08 08:59:56 | [feed][c]Max range AIR: 350.0nm
[fr24feed] 2024-03-08 08:59:56 | [feed][c]Max range GND: 80.0nm
[fr24feed] 2024-03-08 08:59:56 | [feed][i]configuration changed
[fr24feed] 2024-03-08 08:59:56 | [feed][i]defined 3 servers
[fr24feed] 2024-03-08 08:59:56 | [feed][c]Timestamps: optional
[fr24feed] 2024-03-08 08:59:56 | 24-03-08 09:00:23.406 [I][receiver_ac_sender.cpp:137] Stopping ReceiverACSender threads for feed 
[fr24feed] 2024-03-08 08:59:56 | 24-03-08 09:00:23.406 [D][receiver_ac_sender.cpp:141] Stop called on non-running network thread for feed 
[fr24feed] 2024-03-08 08:59:56 | 24-03-08 09:00:23.406 [I][receiver_ac_sender.cpp:96] Configured ReceiverACSender: xxx:8099,xxx:8099,xxx:8099, feed: xxx, send_interval: 5s/1s, max age: 15s, send metadata: true, mode: 1, filtering: true
[fr24feed] 2024-03-08 08:59:56 | 24-03-08 09:00:23.406 [I][crxstats.cpp:588] [stats]Stats thread started
[fr24feed] 2024-03-08 08:59:56 | 24-03-08 09:00:23.406 [I][receiver_ac_sender.cpp:36] Network thread connecting to xxx:8099 for feed xxx
[fr24feed] 2024-03-08 08:59:56 | [feed][n]xxx@xxx:8099/UDP
[fr24feed] 2024-03-08 08:59:56 | [feed][n]connecting
[fr24feed] 2024-03-08 08:59:56 | [feed][n]connected via UDP (fd 23)
[fr24feed] 2024-03-08 08:59:56 | [feed][n]working
[dump1090] rtlsdr: tuner gain set to 49.6 dB (gain step 28)
[dump1090] Allocating 4 zero-copy buffers
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service radarbox: stopping
s6-rc: info: service planefinder-log: stopping
s6-rc: info: service piaware: stopping
s6-rc: info: service opensky: stopping
s6-rc: info: service netcat: stopping
s6-rc: info: service http: stopping
s6-rc: info: service fr24feed: stopping
s6-rc: info: service adsbhub: stopping
s6-rc: info: service adsbfi-mlat: stopping
s6-rc: info: service adsbexchange-stats: stopping
s6-rc: info: service adsbexchange-mlat: stopping
s6-rc: info: service planefinder-log successfully stopped
s6-rc: info: service planefinder: stopping
s6-rc: info: service http successfully stopped
s6-rc: info: service init-upintheair: stopping
s6-rc: info: service confd-layers: stopping
s6-rc: info: service confd-html: stopping
s6-rc: info: service init-upintheair successfully stopped
s6-rc: info: service confd-html successfully stopped
s6-rc: info: service radarbox successfully stopped
s6-rc: info: service conf-radarbox: stopping
s6-rc: info: service opensky successfully stopped
s6-rc: info: service conf-opensky: stopping
s6-rc: info: service piaware successfully stopped
s6-rc: info: service confd-piaware: stopping
s6-rc: info: service netcat successfully stopped
s6-rc: info: service fr24feed successfully stopped
s6-rc: info: service confd-fr24feed: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service adsbhub successfully stopped
s6-rc: info: service conf-adsbhub: stopping
s6-rc: info: service adsbfi-mlat successfully stopped
s6-rc: info: service adsbfi-feed: stopping
s6-rc: info: service adsbexchange-mlat successfully stopped
s6-rc: info: service planefinder successfully stopped
s6-rc: info: service conf-planefinder: stopping
s6-rc: info: service confd-layers successfully stopped
s6-rc: info: service conf-radarbox successfully stopped
s6-rc: info: service conf-opensky successfully stopped
s6-rc: info: service confd-piaware successfully stopped
s6-rc: info: service confd-fr24feed successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service conf-adsbhub successfully stopped
s6-rc: info: service adsbexchange-stats successfully stopped
s6-rc: info: service adsbexchange-feed: stopping
s6-rc: info: service conf-planefinder successfully stopped
s6-rc: info: service adsbfi-feed successfully stopped
s6-rc: info: service conf-adsbfi: stopping
s6-rc: info: service adsbexchange-feed successfully stopped
s6-rc: info: service conf-adsbexchange: stopping
s6-rc: info: service dump1090: stopping
s6-rc: info: service conf-adsbfi successfully stopped
s6-rc: info: service conf-adsbexchange successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service dump1090 successfully stopped
```

Thom-x, 2024-03-08:
Is it only with the latest release or [1.30.0](https://hub.docker.com/layers/thomx/fr24feed-piaware/1.30.0/images/sha256-c050e49ef387a8106e5850d2a16e944f4b4a936e6b301b6ebb8ca3aadefd8b3f?context=explore) is OK ?
Could you try running only one service at a time to find the culprit ? https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/blob/master/README.md#common

fl0wfr, 2024-03-08:
Hi Thomas,
I'm using the 1.32.0. It seems to work as soon as I set SERVICE_ENABLE_HTTP to false.
It seems to be the same behaviour as my previous issue #107

Thom-x, 2024-03-08:
Can you try with https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/107#issuecomment-1516311871 ?

fl0wfr, 2024-03-08:
It works, thank you!

closed 2024-03-08 by Thom-x