Thom-x/docker-fr24feed-piaware-dump1090 #163: Address already in use, container stops

sneak-it, 2024-03-20:
Just a backstory: I have been running this container without issue for about a week, and I took it down to add more feeding destinations. Once I booted the container back up, I have had no luck getting it to stay up.

## Specifications

  - Image hash or version: 1.32.0, latest
  - Platform: Linux a la Synology flavor
  - Subsystem: x86_64
  - Docker: 20.10.23

## Expected Behavior
FR24Feeder + Dump1090 to play nicely.

## Actual Behavior
FR24Feeder causes container to stop. 
`[31m[dump1090][0m Error opening the listening port 30003 (Basestation TCP output): bind: Address already in use`

## Steps to Reproduce the Problem

Things I've done to troubleshoot:
On the host: `netstat -tulpn` does not show port 30003 in use, to confirm `sudo netstat -pna | grep 30003` results in nothing.
Entirely wiped the project, deleted the image, re-downloaded image and re-deployed.
Disabled all services except Dump1090 - this boots. Once FR24Feed is enabled alongside Dump1090, have the issue below.

What confuses me is that no system/docker updates were done, no configurations were changed, simply restarted the container and now it absolutely refuses to boot.

Logs:
```
2024/03/20 01:00:19	stderr	s6-rc: info: service s6rc-oneshot-runner successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service dump1090 successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service s6rc-oneshot-runner: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-planefinder successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbexchange successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbfi successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service dump1090: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbexchange: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbexchange-feed successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbfi: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbfi-feed successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbhub successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-fr24feed successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-html successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-layers successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service init-upintheair successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-opensky successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-piaware successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-planefinder: stopping
2024/03/20 01:00:19	stdout	[35m[adsbfi-mlat][0m Listening for Beast-format results connection on port 30179
2024/03/20 01:00:19	stdout	[35m[adsbfi-mlat][0m Listening for Basestation-format results connection on port 31009
2024/03/20 01:00:19	stdout	[35m[adsbfi-mlat][0m mlat-client 0.4.2 starting up
2024/03/20 01:00:19	stderr	s6-rc: info: service planefinder successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-radarbox successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service fix-attrs successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbexchange-feed: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbexchange-mlat successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbexchange-stats successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbfi-feed: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbfi-mlat successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-adsbhub: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service adsbhub successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-fr24feed: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service fr24feed successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-html: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-layers: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service init-upintheair: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service http successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service netcat successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service conf-opensky: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service opensky successfully stopped
2024/03/20 01:00:19	stderr	s6-rc: info: service confd-piaware: stopping
2024/03/20 01:00:19	stderr	s6-rc: info: service piaware successfully stopped
[... 205 lines cut ...]
    devices:
      - "/dev/bus/usb/001"
    # ports
    # 8754 - DUMP1090 server port
    # 8080 - HTTP server port
    # 30003 - BS Decoded Data
    # 30005 - BEAST Host IP
    # 30105 - MLAT Server
    ports:
      - "8754:8754"
      - "8082:8080"
      - "30003:30003"
      - "30005:30005"
      - "30105:30105"
```

With dump1090+piaware running, inside of the container:
```
root@691f527866b7:/# netstat -pna | grep 30003                            
tcp        0      0 0.0.0.0:30003           0.0.0.0:*               LISTEN      99/dump1090                                                                       
tcp6       0      0 :::30003                :::*                    LISTEN      99/dump1090
```
Should dump1090 be listening on 30003?

I have been successfully feeding to FlightAware, FR24, ADSBFI, ADSBHUB for a few days, when I went to add ADSB Exchange, this is where I started running into problems. I'm willing to bet this is a problem on my end, I just can't quite seem to figure out what the issue is.

sneak-it, 2025-03-31:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Still an issue.

bantzorg, 2025-11-26:
Hey there - ran into this same problem today when 

> I went to add ADSB Exchange, this is where I started running into problems

As soon as I changed to `- "SERVICE_ENABLE_ADSBEXCHANGE=false"` everything started working again.