Thom-x/docker-fr24feed-piaware-dump1090 #171: Container keep crashing

davidepanato, 2024-05-16:
## Specifications

  - Image hash or version: 1.32.1
  - Platform: raspberry pi 4, ubuntu server 24.04
  - Subsystem: ARM
  -  Docker: version 26.1.2, build 211e74b

## Expected Behavior
container runs without problems

## Actual Behavior
container stops, issue seems linked to container unable to access rtl-sdr v3 usb pendrive
another container created with ubuntu can directly access the device without any problem (tested with sdr-test)

## Steps to Reproduce the Problem

  1. configure the system via docker compose
  `services:
  fr24feed-piaware:
    image: "thomx/fr24feed-piaware"
    container_name: fr24feed-piaware
    restart: unless-stopped
    volumes:
      - /etc/localtime:/etc/localtime:ro # needed to sync time for MLAT
    environment:
      - "FR24FEED_FR24KEY=xxxxxxxxxxxxxxx"
      - "SERVICE_ENABLE_FR24FEED=true"
      - "SERVICE_ENABLE_DUMP1090= true"
      - "SERVICE_ENABLE_PIAWARE=false"
      - "SERVICE_ENABLE_HTTP=false"
      - "HTML_SITE_LAT=45.0632"
      - "HTML_SITE_LON=7.7113"
      - "HTML_SITE_ALT=250"
      - "HTML_SITE_NAME=Radar-Dave"
      - "HTML_DEFAULT_TRACKER=Flightradar24"
      - "HTML_RECEIVER_STATS_PAGE_FLIGHTAWARE=https://flightaware.com/adsb/stats/user/"
      - "HTML_RECEIVER_STATS_PAGE_FLIGHTRADAR24=https://www.flightradar24.com/account/feed-stats/?id=xxxxx"
      - "HTML_FR24_FEEDER_STATUS_PAGE=http://192.168.xxx.xxx:8754/"
    devices:
      - "/dev/bus/usb"
    # ports
    # 8754 - DUMP1090 server port
    # 8080 - HTTP server port
    ports:
      - "8754:8754"
      - "8080:8080"
    logging:
      options:
        max-size: "30m"
    tmpfs:
      - /run:exec,size=32M
      - /planefinder/log:exec,size=32M
      - /usr/lib/fr24/public_html/data:size=32M`
  2. launch the container via `docker compose up`
  3. container crash with the following log:
  `docker compose up

[+] Running 1/1
 ✔ Container fr24feed-piaware  Recreated                                                                                                                                                                          0.3s
Attaching to fr24feed-piaware
fr24feed-piaware  | s6-rc: info: service dump1090: starting
fr24feed-piaware  | s6-rc: info: service s6rc-oneshot-runner: starting
fr24feed-piaware  | s6-rc: info: service dump1090 successfully started
fr24feed-piaware  | s6-rc: info: service netcat: starting
fr24feed-piaware  | s6-rc: info: service s6rc-oneshot-runner successfully started
fr24feed-piaware  | s6-rc: info: service fix-attrs: starting
fr24feed-piaware  | s6-rc: info: service init-upintheair: starting
fr24feed-piaware  | s6-rc: info: service confd-piaware: starting
fr24feed-piaware  | s6-rc: info: service confd-layers: starting
fr24feed-piaware  | s6-rc: info: service confd-html: starting
[... 93 lines cut ...]
fr24feed-piaware  | s6-rc: info: service adsbfi-mlat successfully stopped
fr24feed-piaware  | s6-rc: info: service adsbfi-feed: stopping
fr24feed-piaware  | s6-rc: info: service fr24feed successfully stopped
fr24feed-piaware  | s6-rc: info: service confd-fr24feed: stopping
fr24feed-piaware  | s6-rc: info: service adsbexchange-mlat successfully stopped
fr24feed-piaware  | s6-rc: info: service adsbexchange-feed: stopping
fr24feed-piaware  | s6-rc: info: service conf-adsbhub successfully stopped
fr24feed-piaware  | s6-rc: info: service conf-opensky successfully stopped
fr24feed-piaware  | s6-rc: info: service init-upintheair successfully stopped
fr24feed-piaware  | s6-rc: info: service confd-layers successfully stopped
fr24feed-piaware  | s6-rc: info: service confd-html successfully stopped
fr24feed-piaware  | s6-rc: info: service confd-fr24feed successfully stopped
fr24feed-piaware  | s6-rc: info: service planefinder successfully stopped
fr24feed-piaware  | s6-rc: info: service conf-planefinder: stopping
fr24feed-piaware  | s6-rc: info: service conf-planefinder successfully stopped
fr24feed-piaware  | s6-rc: info: service adsbfi-feed successfully stopped
fr24feed-piaware  | s6-rc: info: service conf-adsbfi: stopping
fr24feed-piaware  | s6-rc: info: service adsbexchange-feed successfully stopped
fr24feed-piaware  | s6-rc: info: service conf-adsbexchange: stopping
fr24feed-piaware  | s6-rc: info: service dump1090: stopping
fr24feed-piaware  | s6-rc: info: service conf-adsbfi successfully stopped
fr24feed-piaware  | s6-rc: info: service dump1090 successfully stopped
fr24feed-piaware  | s6-rc: info: service conf-adsbexchange successfully stopped
fr24feed-piaware  | s6-rc: info: service s6rc-oneshot-runner: stopping
fr24feed-piaware  | s6-rc: info: service s6rc-oneshot-runner successfully stopped`

pagottoo, 2024-06-02:
same issue here... recently upgrade my raspberry pi k8s cluster, to ubuntu 24.04, and started to face this problem...

harenber, 2024-07-13:
Just updated my x86 based server running Alpine 3.19.2 (kernel update to 6.6.39-0-lts) and facing now the same problem.
Already tried to unload the `rtl2832_sdr` kernel module - just in case it's in the way -.

closed 2025-04-08 by github-actions[bot]