Thom-x/docker-fr24feed-piaware-dump1090 #172: Support of RTL-SDR v4

leowongco, 2024-05-29:
I have the issue of feeding signal on RTL-SDR, I followed the instruction to install driver on my ubuntu, but there is no signal (only 1-2 msgs on ADS-B), its the docker supported on this dongle, if not, can you add the driver on the docker? thanks.
[image]
[image]
[image]
[image]

jaimealex, 2024-10-01:
I'm in the same situation.
I couldn't configure it in any way, it doesn't receive any ADS-B data.

Thom-x, 2025-03-31:
@leowongco If you are still willing to try, I have updated rtl-sdr, so it might work now.

kamyweb, 2025-10-05:
The same for me - I add on my closed issue some logs.
I didn't undestand why on my laptop the same stick with the same docker config works correclty - it seems Dockerfile on raspberry do something wrong.