Thom-x/docker-fr24feed-piaware-dump1090 #173: Can't connect to my Realtek RTL2838UHIDIR

tzippy84, 2024-07-26:
Hello!

I have set up this docker through a Home Assistant Add-On.

I installed the Add-On, plugged in the USB device into the Intel NUC and rebooted the NUC.
Eventhough hassio can detect the device on the USB bus, I think the Log of the docker states that it cannot connect to the receiver:

[fr24feed] 2024-07-26 14:58:32 | [reader][i]Connecting to unknown receiver via (tcp://127.0.0.1:30005)
[fr24feed] 2024-07-26 14:58:32 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://127.0.0.1:30005

How can I get the Add-On to recognize the device?

Thom-x, 2024-07-26:
You should get the logs from dump1090, fr24feed is just trying to connect to it.
So better see the root cause here.
You can try to only enable dump1090.

tzippy84, 2024-07-26:
I think my mistake was to disable Dump1090 alltogether in the Add-On Settings.
Its working now after enabling that. 
Thank you for the super-fast reply!

Thom-x, 2024-07-26:
Cool ! Closing then 😊

closed 2024-07-26 by Thom-x