Thom-x/docker-fr24feed-piaware-dump1090 #174: Setup Assistance -- Define RTL-SDR to use

heyhewmike, 2024-09-11:
## Specifications

  - Image hash or version: 1.32.1
  - Platform: Home Assistant - https://github.com/MaxWinterstein/homeassistant-addons/
  
I am going to be using 2 identical RTL-SDR units. For rtlamr2mqtt I am required/able to define the unit/device to use when there are multiple SDRs.

Is there the ability to define which of the 2 SDRs to use?
Will the platform be able to identify which one is free?
Do I need to ensure that my rtlamr2mqtt is running first so that this does not start the 'wrong' one?

Thom-x, 2024-09-11:
Instead of 

```
	--device=/dev/bus/usb:/dev/bus/usb \
```
You can pass the exact usb device you want to use.

heyhewmike, 2024-09-11:
Thank you. It appears I am now stuck between you and the addon I linked to above. It also appears I am stuck due to the issue that I have 2 identical devices with the same serial and device IDs.

I have closed the issue as you have provided the answer to the question and request for assistance.

closed 2024-09-11 by heyhewmike