Thom-x/docker-fr24feed-piaware-dump1090 #175: How to locate RTL-SDR v5 on macOS for the docker device ID /dev/?????

PlasmaSoftUK, 2024-09-14:
## Specifications

  - Image hash or version:  latest
  - Platform:   macOS Sequoia
  - Subsystem:  Docker

## Expected Behavior

WIth the R~TL-SDR v5 plugged in there is no device listed in /dev that I can use to tell dump1090 where the SDR is. It is there and can be found by SDR++ and also when I do a  'system_profiler SPUSBDataType'  :

`    USB 3.0 Bus:

      Host Controller Driver: AppleEmbeddedUSBXHCIFL1100
      PCI Device ID: 0x1100 
      PCI Revision ID: 0x0010 
      PCI Vendor ID: 0x1b73 

        NESDR SMArt v5:

          Product ID: 0x2838
          Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
          Version: 1.00
          Serial Number: 00000001
          Speed: Up to 480 Mb/s
          Manufacturer: Nooelec
          Location ID: 0x02200000 / 3
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0`

## Actual Behavior

Unable to locate a device so container won't start

## Steps to Reproduce the Problem

  1.  Create a docker container
  2. Fill in my ID for FlightRadar
  3. Unable to enter a device ID for dump1080 as no device for the SDR listed in /dev

PlasmaSoftUK, 2024-09-15:
I realise I should have posted this on the dump1090, I will go do that now, but if anyone has any ideas how to fix this that would be great.

Engineero, 2024-10-18:
Bumping this. I'd love to see a solution and if I figure something out I'll report back.

closed 2025-04-08 by github-actions[bot]