Thom-x/docker-fr24feed-piaware-dump1090 #179: Could not connect to the GPS receiver

yayitazale, 2024-11-22:
## Specifications

  - Image hash or version: 1.32.1

## Expected Behavior

I'm getting this error constantly:
[gps][w]Could not connect to the GPS receiver

## Actual Behavior

I have a USB GPS receiver attached to RPI and I can see that it is getting position executing"cgps -s":

[image]

[image]

I have "--device=/dev/bus/usb:/dev/bus/usb" so container should see the GPS receiver too.

I'm doing something wrong?

Thom-x, 2025-03-31:
@yayitazale If you are still willing to try, I have updated rtl-sdr and other feeder, so it might work now.

yayitazale, 2025-04-01:
I will try as soon as possible and let you know, thanks!

yayitazale, 2025-04-01:
Looks good! Thank you!

closed 2025-04-01 by yayitazale