Thom-x/docker-fr24feed-piaware-dump1090 #181: seems directory piaware.conf is conflicting with creation of file piaware.conf

ibigsnet, 2025-02-05:
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO Backend set to env
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO Starting confd
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO Backend source(s) set to 
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO /etc/piaware.conf has mode drw-r--r-- should be -rw-r--r--
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO /etc/piaware.conf has md5sum d41d8cd98f00b204e9800998ecf8427e should be 2406589a69a744d51d1bdaef8c27323a
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: INFO Target config /etc/piaware.conf out of sync
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: ERROR rename /etc/.piaware.conf501596390 /etc/piaware.conf: file exists
2025-02-05T13:02:24-05:00 c8721871badc /opt/confd/bin/confd[112]: FATAL rename /etc/.piaware.conf501596390 /etc/piaware.conf: file exists
s6-rc: warning: unable to start service confd-piaware: command exited 1

root@c8721871badc:/etc# ls -lah | grep pi*
drw-r--r-- 1 root root     38 Feb  5 12:56 piaware.conf
root@c8721871badc:/etc#

Why is this building a directory, then trying to write to it as a file?

ibigsnet, 2025-02-05:
Changing to false for all services / service flags:

Log file:
s6-rc: info: service dump1090: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service dump1090 successfully started
s6-rc: info: service netcat: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service init-upintheair: starting
s6-rc: info: service confd-piaware: starting
s6-rc: info: service confd-layers: starting
s6-rc: info: service confd-html: starting
s6-rc: info: service confd-fr24feed: starting
s6-rc: info: service conf-radarbox: starting
s6-rc: info: service conf-planefinder: starting
s6-rc: info: service conf-opensky: starting
s6-rc: info: service conf-adsbhub: starting
s6-rc: info: service conf-adsbfi: starting
s6-rc: info: service conf-adsbexchange: starting
s6-rc: info: service netcat successfully started
s6-rc: info: service confd-layers successfully started
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service init-upintheair successfully started
s6-rc: info: service confd-piaware successfully started
s6-rc: info: service piaware: starting
s6-rc: info: service confd-html successfully started
s6-rc: info: service http: starting
s6-rc: info: service confd-fr24feed successfully started
s6-rc: info: service fr24feed: starting
s6-rc: info: service conf-radarbox successfully started
s6-rc: info: service radarbox: starting
s6-rc: info: service conf-planefinder successfully started
s6-rc: info: service planefinder: starting
s6-rc: info: service conf-opensky successfully started
s6-rc: info: service opensky: starting
s6-rc: info: service conf-adsbhub successfully started
s6-rc: info: service adsbhub: starting
s6-rc: info: service conf-adsbfi successfully started
s6-rc: info: service adsbfi-feed: starting
s6-rc: info: service conf-adsbexchange successfully started
s6-rc: info: service adsbexchange-feed: starting
s6-rc: info: service piaware successfully started
s6-rc: info: service http successfully started
s6-rc: info: service fr24feed successfully started
s6-rc: info: service radarbox successfully started
s6-rc: info: service planefinder successfully started
s6-rc: info: service planefinder-log: starting
s6-rc: info: service opensky successfully started
s6-rc: info: service adsbhub successfully started
s6-rc: info: service adsbfi-feed successfully started
s6-rc: info: service adsbfi-mlat: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service adsbexchange-feed successfully started
s6-rc: info: service adsbexchange-stats: starting
s6-rc: info: service adsbexchange-mlat: starting
s6-rc: info: service planefinder-log successfully started
s6-rc: info: service adsbfi-mlat successfully started
s6-rc: info: service adsbexchange-stats successfully started
s6-rc: info: service adsbexchange-mlat successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started

Host machine:
root@KRYZEN:/mnt/user/appdata/fr24feed-piaware# ls -lah
total 0
drwxrwxrwx 1 root   root    0 Feb  5 13:11 ./
drwxrwxrwx 1 nobody users 442 Feb  5 13:07 ../
root@KRYZEN:/mnt/user/appdata/fr24feed-piaware#

No files are found. Seems file creation is all temporary and never rights to the host OS running the Docker Container.

Doublechecking my mounts:
root@KRYZEN:/mnt/user/appdata/fr24feed-piaware# docker inspect fr24feed-piaware | grep -i "Mounts"
        "Mounts": [

Kinda odd I think... but my XML should be doing the path flag:
<Config Name="Piaware Config" Target="/etc/piaware.conf" Default="" Mode="rw" Description="" Type="Path" Display="always" Required="true" Mask="false">/mnt/user/appdata/fr24feed-piaware/</Config>

Command execution
docker run
  -d
  --name='fr24feed-piaware'
  --net='br0'
  --ip='10.1.0.22'
  --pids-limit 2048
  -e TZ="America/New_York"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="KRYZEN"
  -e HOST_CONTAINERNAME="fr24feed-piaware"
  -e 'TCP_PORT_8080'='8080'
  -e 'SERVICE_ENABLE_DUMP1090'='false'
  -e 'SERVICE_ENABLE_HTTP'='false'
  -e 'SERVICE_ENABLE_FR24FEED'='false'
  -e 'SERVICE_ENABLE_PIAWARE'='true'
  -e 'PIAWARE_FEEDER_DASH_ID'='39a1d7a1-beb7-4e4e-8313-7260517b70e5'
  -e 'PIAWARE_RECEIVER_DASH_TYPE'='other'
  -e 'PIAWARE_RECEIVER_DASH_HOST'='127.0.0.1'
  -e 'TCP_PORT_PIAWARE_RECEIVER_DASH_PORT'='30005'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='http://[IP]:[PORT:8080]/gmap.html'
  -l net.unraid.docker.icon='https://github.com/FoxxMD/unraid-docker-templates/raw/master/icons/flightaware.jpg'
  -v '/mnt/user/appdata/fr24feed-piaware/':'/etc/piaware.conf':'rw'
  -v '/mnt/user/appdata/fr24feed-piaware/':'/usr/lib/fr24/public_html/config.js':'rw'
  -v '/mnt/user/appdata/fr24feed-piaware/':'/etc/fr24feed.ini':'rw'
  --device='/dev/bus/usb/003/006' 'thomx/fr24feed-piaware:latest'
c35ccc842dc291066bbcade3c5d2da83f02924e44e7c63fb4f6a697aa619ed69

The command finished successfully!

If I change -v '/mnt/user/appdata/fr24feed-piaware/':'/etc/piaware.conf':'rw' to -v '/mnt/user/appdata/fr24feed-piaware/':'/etc':'rw'
this doesn't work either. So something to do with the startup process of the container on a fresh install.

Thom-x, 2025-02-05:
> If I change -v '/mnt/user/appdata/fr24feed-piaware/':'/etc/piaware.conf':'rw' to -v '/mnt/user/appdata/fr24feed-piaware/':'/etc':'rw'
this doesn't work either. So something to do with the startup process of the container on a fresh install.

Don't mount any file, use env var only

Thom-x, 2025-02-05:
You can mount the specific log file if you want, but not the config used by env var.

closed 2025-03-29 by Thom-x